lvgl/lvgl #319: Bug when style of 'ta' set up "lv_ta_set_one_line"

as564335sa, 2018-07-20:
After call "lv_ta_set_one_line",the object “ta” can still input "Enter",I think it's Incorrect.

kisvegabor, 2018-07-22:
Hi,

I tested it with this:
```c
    lv_obj_t * ta = lv_ta_create(lv_scr_act(), NULL);
    lv_ta_set_one_line(ta, true);
    lv_ta_set_text(ta, "a\nb");
```

And got: 
[image]

So the most simple code above worked correctly. Can you provide a code snippet which shows a not -working scenario?

as564335sa, 2018-07-23:
Hi,
My code is the same as yours，The difference is that I use external keyboard or “kb” object to input "Enter" instead of using "lv_ta_set_text" API;
This my code:
`lv_obj_t *ta;
	ta = lv_ta_create(cont,NULL);
	lv_ta_set_one_line(ta,true);
	lv_obj_align(ta,cont,LV_ALIGN_CENTER,0,0);
	lv_obj_set_size(ta,200,20);
	lv_ta_set_style(ta,LV_TA_STYLE_BG,&cont_style);
	lv_ta_set_text(ta,"0");
	lv_group_add_obj(group,ta);`

kisvegabor, 2018-07-25:
Hi,

Finally, I find the bug. It was more complicated than it seems because LittlevGL tried to ignore line breaks in the case of Text area but - as you can see - it failed in some cases. 

Now I simplified it and `\n` characters are just ignored in `lv_ta_add_char`.

Please verify if it works!

closed 2018-08-02 by as564335sa