lvgl/lvgl #324: #include "../../lv_abc.h" - remove the '..'

aenchevich, 2018-07-23:
Let's remove all relative include reference, please?
Ultimately the lib and headers should be installed to correct locaitons on target so these are not needed and confuse the build tools.

kisvegabor, 2018-07-23:
Hi,

If the relative include path would be removed then at least the `lvgl`folder should be added as include path. However, not all IDE/toolchain supports to define additional include paths. For example, in Arduino IDE there is no option for this.

Sidenote:
In `dev-5.2` there will an improvement to make possible to freely place `lv_conf.h`  https://github.com/littlevgl/lvgl/issues/294

aenchevich, 2018-07-26:
Ok so, yes - it is really the lv_[...]conf.h includes that mess things up with their "../" paths. 

The problem I am facing was:
- I am building a lib (liblittlevgl.a and liblittlevgl.so), 
- then placing the lib(s) at the standard lib location (e.g. /usr/lib)
- then placing the headers also at the standard include location:  /usr/include/littlevgl/lvgl.
- The lv_[...]conf.h files are NOT placed into /usr/include/littlevgl/ because this would make the configuration global for all projects using lvgl. 

So the build breaks because various files are trying to locate "lv_conf.h" or "../lv_conf.h" or "../../lv_conf.h"

The option in dev-5.2 to specify the location of lv_conf.h will probably solve this issue, which is great! 

However I am still strongly advocating that all the sources that need lv_conf.h must include it as:
`#include "lv_conf.h"`

Having ../ or ../../ (and even ../../../ in one of the files) creates the potential for build problems of at least two types:
a) what I described above (unable to find the file)
b) the possibility of including different lv_conf.h files which could generate build inconsistencies and hard to find bugs. Consider this build environment: 

`  CFLAGS = -I/dir/1/2/3      -I/dir/1/2        -I/dir/1`

Header files :
```
/dir/1/2/lv_conf.h
/dir/1/lv_conf.h
/dir/lv_conf.h
```

Now, depending on which of the 3 include variants :

```
#include "lv_conf.h"
#include "../lv_conf.h"
#include "../../lv_conf.h"
```

 are used in your source you will end up with 3 different lv_conf.h files used in your build!
 - - - -
Regarding the Arduino IDE... well Arduino is known for hacky non-standard solutions... nevertheless we don't want to break it of course... hmmm.

I downloaded the IDE and took a quick look at some parts of it - there is a file named boards.txt which seems to be used (among other things) to pass compilation options per-board? Can it be used for this? Or maybe there is some other way to do this in the Arduino IDE after all (I admit I am not familiar with this environment...)?

kisvegabor, 2018-07-27:
Hi,

Have look at this: https://github.com/littlevgl/lvgl/blob/dev-5.2/lv_misc/lv_mem.h#L17
So you need to add `-DLV_CONF_INCLUDE_SIMPLE` and `-Ipath/to/conf`.

For me, it seems it solves your issue. And it also keeps the current relative include method for simpler use cases and environemnts.

What do you think?

aenchevich, 2018-07-27:
I think 
a)  that what you added in 5.2 is completely adequate to solve my immediate issue (thank you!) ;
b)  that the proper way would be to make it work with #include "lv_conf.h" ;
and c) I am sure that you have many more important things to work on, so this will do for the moment and I will close the issue now! 

Thanks!
:)

closed 2018-07-27 by aenchevich

aenchevich, 2018-08-02:
I want to ask a question, it is related to the problem described in this issue ("../.." include paths) and it applies to lv_drivers. 

I did not want to create a new issue in lv_drivers just to ask the question so I am asking here, hope it's ok. Not sure if I need to re-open this issue to get your attention...
- - - - -
These files in lv_drivers have similar includes (which cause similar problems):

```
display/R61581.h:16:#include "../../lv_drv_conf.h"
display/SSD1963.h:16:#include "../../lv_drv_conf.h"
display/ST7565.h:16:#include "../../lv_drv_conf.h"
display/fbdev.h:16:#include "../../lv_drv_conf.h"
display/monitor.h:16:#include "../../lv_drv_conf.h"
indev/mouse.h:17:#include "../../lv_drv_conf.h"
indev/keyboard.h:16:#include "../../lv_drv_conf.h"
indev/AD_touch.h:17:#include "../../lv_drv_conf.h"
indev/XPT2046.h:16:#include "../../lv_drv_conf.h"
indev/FT5406EE8.h:16:#include "../../lv_drv_conf.h"
```
Do plan to fix this (perhaps re-using LV_CONF_INCLUDE_SIMPLE or renaming it to something like LV_INCLUDES_SIMPLE or LV_INCLUDES_LOCAL?)...

Would you merge a pull req. if I fixed this?

Thanks
:)

kisvegabor, 2018-08-04:
Hi,

Yes, `LV_CONF_INCLUDE_SIMPLE` should be applied there too. I Pull Request would be welcome! :)