lvgl/lvgl #327: Infinite loop bug with lists.

hydra, 2018-07-24:
This code causes an infinite loop:

```c
    lv_obj_t *list = lv_list_create(panel, NULL);
    lv_obj_clean(list);
    lv_list_add(list, SYMBOL_EDIT, "TEST", NULL);
```

stackstrace:
```
Thread #1 57005 (Suspended : Signal : SIGTRAP:Trace/breakpoint trap)	
	lv_obj_get_screen() at lv_obj.c:1,057 0x801118a	
	lv_obj_invalidate() at lv_obj.c:357 0x801067c	
	lv_obj_set_parent() at lv_obj.c:436 0x80107d8	
	lv_page_signal() at lv_page.c:469 0x801ecf6	
	lv_list_signal() at lv_list.c:495 0x801df96	
	lv_obj_create() at lv_obj.c:258 0x8010528	
	lv_cont_create() at lv_cont.c:67 0x801a61e	
	lv_btn_create() at lv_btn.c:57 0x8019e46	
	lv_list_add() at lv_list.c:155 0x801dbd2	
	userCode() at userCode.c:57 0x802e9f6	
	<...more frames...>	
```

the infinite loop in question is this:

https://github.com/littlevgl/lvgl/blob/master/lv_core/lv_obj.c#L1045-L1049

I encountered this issue when attempting to remove old and add new buttons to a list view object.

What I want to do this this:

```c
void createElements() {
    lv_obj_t *list = lv_list_create(panel, NULL);
}
```

then at any random time later on in the life of the program

```c
void refreshElements() {
    lv_obj_clean(list);
    lv_list_add(list, SYMBOL_EDIT, "some element name", someHandler);
    lv_list_add(list, SYMBOL_EDIT, "some element name", someHandler);
    // ...
}
```

hydra, 2018-07-24:
for the record, using `lv_ovj_del(item)` on the list items works instead of `lv_ovj_clean(list)`, but I don't want/need to keep a reference to each item (lv is already doing that for me...).

I'm feeling that the list API needs a `lv_list_remove_all()` method.

kisvegabor, 2018-07-25:
Hi,

See: https://github.com/littlevgl/lvgl/issues/161

In summary: In `dev-v5.2` there are functions (`lv_..._clean()`) to clean the content of the objects where `lv_obj_clean` is not working. So you should use `lv_list_clean` for this purpose.

hydra, 2018-07-25:
ok, I missed #161 when I did a search for other bug reports, sorry for duplicate.

closed 2018-07-25 by hydra

kisvegabor, 2018-07-25:
No problem! :)