lvgl/lvgl #328: A strange problem about "group"

as564335sa, 2018-07-24:
Hi,
I encountered a strange problem when using external keyboard and group on "STM32F103";
My logic :
1、Create a group;
2、Create "window1",and create a hidden button on it，and set the button click event,add "btn" to group
3、When press "enter" key , delete "window1" to save rams，and create "window2" at the same time；
4、Create some objects on "window2" and add to group;
5、Create a hidden button on "window2". When press "enter" key, delete "window2" and Create "window1"；

My problem is in “window2”，Part of the code is as follows:
`

lv_res_t makeUpDlg(lv_obj_t *obj)
{
	//printf("pvSetDlg\r\n");
	setCurrentUi(0);
	reloadGroup();
	delPreDlg(obj);

	lv_obj_t *cont;
	cont = lv_cont_create(lv_scr_act(),NULL);
	lv_cont_set_style(cont,&cont_style);
	lv_obj_set_size(cont,LV_HOR_RES,LV_VER_RES);

	setEscCallback(cont,setDlg);

	lv_obj_t *label1;
	label1 = lv_label_create(cont,NULL);
	lv_label_set_text(label1,"设  置 >> 修  正");
	lv_obj_align(label1,cont,LV_ALIGN_IN_TOP_LEFT,30,20);

	lv_obj_t *btn1[6];
	lv_obj_t *label[6];
	for(uint8_t i = 0; i < 6; i ++)
	{
		btn1[i] = lv_btn_create(cont,NULL);
		lv_btn_set_style(btn1[i],LV_BTN_STATE_REL,&cont_style);
		lv_btn_set_style(btn1[i],LV_BTN_STATE_PR,&cont_style);
		lv_btn_set_layout(btn1[i],LV_LAYOUT_OFF);
		lv_obj_set_size(btn1[i],320,34);
		lv_group_add_obj(group,btn1[i]);
		label[i] = lv_label_create(btn1[i],NULL);
	}
	lv_label_set_text(label[0],"1:小信号切除");
	lv_label_set_text(label[1],"2:零点修正");
	lv_label_set_text(label[2],"3:满度修正");
	lv_label_set_text(label[3],"4:总量清零初始设置");
	lv_label_set_text(label[4],"5:总量清零许可");
	lv_label_set_text(label[5],"6:定量累计值自动清零许可");

	lv_obj_align(btn1[0],cont,LV_ALIGN_IN_TOP_MID,-10,60);
	lv_obj_align(btn1[1],btn1[0],LV_ALIGN_OUT_BOTTOM_LEFT,0,0);
	lv_obj_align(btn1[2],btn1[1],LV_ALIGN_OUT_BOTTOM_LEFT,0,0);
	lv_obj_align(btn1[3],btn1[2],LV_ALIGN_OUT_BOTTOM_LEFT,0,0);
	lv_obj_align(btn1[4],btn1[3],LV_ALIGN_OUT_BOTTOM_LEFT,0,0);
	lv_obj_align(btn1[5],btn1[4],LV_ALIGN_OUT_BOTTOM_LEFT,0,0);

	lv_obj_align(label[0],btn1[0],LV_ALIGN_IN_LEFT_MID,10,0);
	lv_obj_align(label[1],btn1[1],LV_ALIGN_IN_LEFT_MID,10,0);
	lv_obj_align(label[2],btn1[2],LV_ALIGN_IN_LEFT_MID,10,0);
	lv_obj_align(label[3],btn1[3],LV_ALIGN_IN_LEFT_MID,10,0);
	lv_obj_align(label[4],btn1[4],LV_ALIGN_IN_LEFT_MID,10,0);
	lv_obj_align(label[5],btn1[5],LV_ALIGN_IN_LEFT_MID,10,0);

	lv_btn_set_action(btn1[0],LV_BTN_ACTION_CLICK,ssignalCut);
	lv_btn_set_action(btn1[1],LV_BTN_ACTION_CLICK,zeroCorrect);
	lv_btn_set_action(btn1[2],LV_BTN_ACTION_CLICK,maxCorrect);
	lv_btn_set_action(btn1[3],LV_BTN_ACTION_CLICK,initTotalVal);
	lv_btn_set_action(btn1[4],LV_BTN_ACTION_CLICK,totalClrEnable);
	lv_btn_set_action(btn1[5],LV_BTN_ACTION_CLICK,fiexTotalClrEnable);

	label1 = lv_label_create(cont,NULL);
	lv_label_set_text(label1,VERSION);
	lv_obj_align(label1,cont,LV_ALIGN_IN_BOTTOM_MID,0,0);

	lv_obj_t *line;
	line = lv_line_create(cont,NULL);
	lv_line_set_points(line,lin_p1,2);
	lv_line_set_style(line,&cont_style);

	line = lv_line_create(cont,line);
	lv_line_set_points(line,lin_p2,2);
	return LV_RES_INV;
}

static lv_res_t maxCorrect(lv_obj_t *obj)
{
	setCurrentUi(10);
	reloadGroup();//Delete group objects and recreate group
	delPreDlg(obj);//Delete the previous window

	lv_obj_t *cont;
	cont = lv_cont_create(lv_scr_act(),NULL);
	lv_cont_set_style(cont,&cont_style);
	lv_obj_set_size(cont,LV_HOR_RES,LV_VER_RES);

	setEscCallback(cont,makeUpDlg);
	
	lv_obj_t *label1;
	label1 = lv_label_create(cont,NULL);
	lv_label_set_text(label1,"满度修正");
	lv_obj_align(label1,cont,LV_ALIGN_IN_TOP_MID,0,20);
	//pos 1:
	
	lv_obj_t *ta;
	ta = lv_ta_create(cont,NULL);
	lv_ta_set_one_line(ta,true);
	lv_obj_align(ta,cont,LV_ALIGN_CENTER,0,0);
	lv_obj_set_size(ta,200,20);
	lv_ta_set_style(ta,LV_TA_STYLE_BG,&cont_style);
	lv_ta_set_text(ta,"0");
	lv_group_add_obj(group,ta);
	
	//pos 2:
	
	label1 = lv_label_create(cont,NULL);
	lv_label_set_text(label1,VERSION);
	lv_obj_align(label1,cont,LV_ALIGN_IN_BOTTOM_MID,0,0);

	lv_obj_t *line;
	line = lv_line_create(cont,NULL);
	lv_line_set_points(line,lin_p1,2);
	lv_line_set_style(line,&cont_style);

	line = lv_line_create(cont,line);
	lv_line_set_points(line,lin_p2,2);
	//pos 3:
	setHiddenEnterBtn(cont,makeUpDlg);
	
	return LV_RES_INV;
}
void setHiddenEnterBtn(lv_obj_t *parent,lv_action_t action)
{
	hidden_enter_btn = lv_btn_create(parent,NULL);
	lv_obj_set_hidden(hidden_enter_btn,true);
	//lv_group_add_obj(group,hidden_enter_btn);
	lv_btn_set_action(hidden_enter_btn,LV_BTN_ACTION_CLICK,action);
}

`
When the function "setHiddenEnterBtn()" is on "//pos 1",the "maxCorrect" window will not be loaded and the program will be circulate in "lv_draw.c"

When the function "setHiddenEnterBtn()" is on "//pos 2".When the "enter" key is pressed, there will be a segment error，I tracked the debug result is when window "makeUpDlg()" delete the "line" object of "maxCorrect()",That's what I can't understand because the others "callback"  of  "makeUpDlg()" It's normal.

I put the "setHiddenEnterBtn()" on "//pos 3:" in the end,and it worked well.I don't know why;
I don't know if I have described the problem clearly . I would like to know if there are any points for attention when using the group?

hydra, 2018-07-24:
please can you edit your post and use github markup to enclose the code in backquotes?

See https://guides.github.com/features/mastering-markdown/#examples

as564335sa, 2018-07-25:
Hi,
I'm sorry！Now I uploaded my project code, running in PC simulator;
The code that appears to be problem is located in "app/pvset.c"，and I added annotations in line.210 and line.222;
The operation steps are：
1、Run the compiled program;
2、Press "enter" key and jump to next "window"
3、Group default focus on "流量"，and press "enter",jupm to next "window"
4、Press "Tab" key to focus on "3:流量上限设定",then the fault will appear;
[pc_simulator_test.zip](https://github.com/littlevgl/lvgl/files/2226678/pc_simulator_test.zip)
Thanks for your help!

kisvegabor, 2018-07-25:
Hi,

As @hydra asked please format the code in your first post using markdown.

I tested your project and I needed to increase `LV_MEM_SIZE` even to see the first screen. I changed `LV_MEM_SIZE` to 8 * 1024 and it worked for me. See the attached image.

Is it working there with larger `LV_MEM_SIZE`?

[image]

as564335sa, 2018-07-26:
Hi,
I don't think it's a memory problem，now i set the `LV_MEM_SIZE` to 100 * 1024 , but the fault still exists;
And it's very strange, see the picture:
The function `setHiddenEnterBtn` is in line.210，it works well;
[image]

The function `setHiddenEnterBtn` is in line.222，and set text of "ta" to "hello" ,it works well;
[image]
But it will get fault when set text of "ta" to others such as "0"、“1”、"123" and so on;
[image]

This is just a part of the problem，sometimes it makes a mistake at different locations of different windows，now I make the program normal by modifying the function position；
But I am not sure whether it is normal after normal use or adding other functions, because I can't fully control the code now . I tried to track debugging, but I can't find the reason now.

as564335sa, 2018-07-26:
Now I've found another problem，see the pictures:
[image]
[image]

kisvegabor, 2018-07-27:
Hi,

It seems like you write to mush data into a buffer or array. Especially according to the last post ('.' changed to ',').

1. Please update lvgl from the `dev-5.2` branch. Thre are some fixes in it, maybe it's a library issue.
2. Run your code in Debug to see where it will halt.

as564335sa, 2018-08-02:
Hi , 
I Debug my code , when this fault is appear:
[image]
The code will break here:
[image]
I really can't understand，can you help me?

as564335sa, 2018-08-02:
Ok，i'm update lvgl from the `dev-5.2` branch，it works well now!

kisvegabor, 2018-08-02:
Huh, great! :)

closed 2018-08-02 by kisvegabor