lvgl/lvgl #329: Ability to rotate the screen at runtime.

hydra, 2018-07-24:
Feature request for the ability to change screen orientation at run-time please.  I can change my display driver and touch screen drivers in run-time just fine so it would be nice to be able to do it with LVGL too.

I feel there's two stages to this:
1) let the user code set orientation for LGL *before* any objects are created.  i.e. around the time you call LVGL's initialization code.
2) let the user code set orientation and have LVGL refresh objects, etc.

I'd like to be able to let uses of my device configure screen orientation and on the next boot have it applied, scenario 1 above takes this into account.

I'm sure other LVGL users would like to have auto-rotation on their devices, scenario 2 above would assist with this.

At the moment neither scenarios are possible, screen orientation is fixed at compile time and this requires different firmware to be created, made available, and uploaded to the device.  Having to create 4 different builds/binaries for this setting is really not ideal!

Zaltora, 2018-07-24:
This  feature is WIP and planned.
i work on it at driver level (new_api branch ) and an API will be added by @kisvegabor to lvgl to handle the screen and input device rotation. 
Discussion about it (at the end): littlevgl/lv_drivers#9
I need do documentation on it. I will get time soon.

Usage will be:
`lv_set_orientation( lv_rotation_t rotation); `

kisvegabor, 2018-07-25:
According to the discussion with @Zaltora this feature will be implemented with the "multi-display support" in v6.0

embeddedt, 2019-01-21:
I'm looking into implementing this fairly soon (hopefully).

I'm wondering how we should handle the variable-length arrays when the size of the display is potentially unbounded.

kisvegabor, 2019-01-21:
@embeddedt It will be a "sub-feature" in multi-display support. As the multi-display support means a huge rework in the internal mechanisms of the library I'd like to do it myself. The orientation changing can be added once the basic updates are done. Because a lot of related issues are opened recently, I will do the basics of multi-display support first in `dev-6.0` to enable the adding of extra features like this.
Is it okay for you?

embeddedt, 2019-01-21:
> @embeddedt It will be a "sub-feature" in multi-display support. As the multi-display support means a huge rework in the internal mechanisms of the library I'd like to do it myself. The orientation [...]

No problem. I'm not too familiar with the low-level drawing code anyways so that is fine with me.

vipin65, 2019-02-03:
Thanks for quick response. Good to know will be available soon. I am planing to modify febdev.c for immediate solution, Is this correct place ?

kisvegabor, 2019-02-04:
You should update [this part](https://github.com/littlevgl/lv_drivers/blob/master/display/fbdev.c#L132) of `fbdev.c` to map the pixels correctly.

amirgon, 2019-02-09:
@kisvegabor Will "Advanced display support" allow setting `LV_HOR_RES` and `LV_VER_RES` on runtime?  

This is very important for Micropython, where the user registers display/input devices in his script on runtime.  

Think of a user who uses some display and now wants to switch to another display with a different resolution. He can change the driver in Micropython script by importing and registering a different driver, but now he would still need to re-build the Micropython project to change the resolution.
Ideally a Micropython user should be able to decide which display to use without the need to compile C at all.  

What about other compile time constants? such as `LV_DPI`, `LV_ANTIALIAS`, `LV_REFR_PERIOD`, `LV_VBD*` macros, `LV_INDEV_*` input device settings etc.? I imagine a user would want to change those on runtime as well once he can switch display/input driver on runtime.

kisvegabor, 2019-02-10:
> Will "Advanced display support" allow setting `LV_HOR_RES` and `LV_VER_RES` on runtime?

Yes with a `lv_disp_set_resolution(disp, 400, 300);` function.

> What about other compile time constants? such as `LV_DPI`, `LV_ANTIALIAS`, `LV_REFR_PERIOD`, `LV_VBD*` macros, `LV_INDEV_*` input device settings etc.? I imagine a user would want to change those on runtime as well once he can switch display/input driver on runtime.

I'm planning to move as much of them to the driver as possible to enable run time configuration.

kisvegabor, 2019-03-02:
For run-time display config change it seems the two most important things are:
1. `lv_disp_set_hor/ver_res()`
2. `lv_disp_set_orientation()`

I'll add them (or with an equivalent API) soon to `dev-6.0`

amirgon, 2019-03-02:
> For run-time display config change it seems the two most important things are:
> 
> 1. `lv_disp_set_hor/ver_res()`
> 2. `lv_disp_set_orientation()`
> 
> I'll add them (or with an equivalent API) soon to `dev-6.0`

@kisvegabor On micropython we would like to allow the user switch display **without re-compiling the C code**, just by registering a different driver.  

For this to work we need runtime configuration for hor/ver res, as you added, but **also color depth**, since different displays have different color depths.  
 
1. Do you have plans to make color depth a runtime configuration as well?
2. Is it possible to configure all of them **per display**, in scenarios of multi-display?

kisvegabor, 2019-03-03:
@amirgon 

> For this to work we need runtime configuration for hor/ver res, as you added, but also color depth, since different displays have different color depths.

I made some experiments to see how difficult it is to implement but I faced some obstacles.

I summarized my though and experiences so far:
1. I assume in 99% of cases you will use one display and you know it in advance. Attaching displays in run-time is a very specific thing.
2. If you are using the LCD controller periphery then you also need know color depth in advance.
3. Now the images are generated will all possible color depths and a define selects the corresponding format based on `LV_COLOR_DEPTH`. With run-time color depth config, we would lose this simple and convenient behavior.
4. If the color depth doesn't match in worst case the colors need to be converted in the flush function. Having a 320x240 display and 160 MHz MCU and assuming [color conversation](https://github.com/littlevgl/lvgl/blob/master/lv_misc/lv_color.h#L280) needs 8 instructions (just guessing) then 320 x 240 x 8 / 160M = ~4 ms. With 25 FPS the refresh period is 40ms. With the conversation, it will be 44ms which means 23 FPS. Still acceptable I think. In addition, the run-time color depth change would lower the performance too because more branches need to be added into the most performance critical functions too. (Don't know how big)

So there two big questions:
1. The user really doesn't know the color depth in advance? 
2. Is it really a use-case to have more large displays simultaneously with different color depth? (when the color depth conversation takes too much time)

amirgon, 2019-03-03:
@kisvegabor  
I have a stronger case for configuring everything on runtime, because micropython is all about doing everything in runtime, while building the firmware image only once.  
For simultaneous multi-display scenarios - probably a weaker case. I'm not sure how common they are, especially with different display types.

> 1. I assume in 99% of cases you will use one display and you know it in advance. Attaching displays in run-time is a very specific thing.

Even if you use one display and you know what it is in advance, There is still a need to configure this **on runtime**. I'll explain why **by an example**:   

---

A micropython user buys/builds a board and a display. This user knows a lot about hardware, but not too much about software, that's why he chose micropython - you don't need to be a C expert to create simple embedded software.   

The user is **using a precompiled image** for his board, which is provided by the board manufacturer or  by someone else. This is common - for example there's prebuilt micropython firmware for many boards [on micropython website](http://micropython.org/download), prebuilt firmware for [Adafruit CircuitPython](https://learn.adafruit.com/welcome-to-circuitpython/installing-circuitpython), some evaluation boards come with flashed micropython already, like [this one](https://www.sparkfun.com/products/14413).   
So let's assume this prebuilt image already contains LitttlevGL, since LitttlevGL becomes so popular and successful!  

Now the user tries to connect his display and load a display driver in micropython. If he is lucky, the prebuilt image matches his display configuration. But if not, he would have to build a firmware image himself with some compile-time configuration settings that matches his screen.

---

> 3\. Now the images are generated will all possible color depths and a define selects the corresponding format based on `LV_COLOR_DEPTH`. With run-time color depth config, we would lose this simple and convenient behavior.

Not necessarily.   
You could use function pointers to configure on runtime which function is called (different function for each color depth).

> 4\. If the color depth doesn't match in worst case the colors need to be converted in the flush function.

That's right. It takes more cycles, but it also means more RAM consumption, right? If color is configured as 32bit but the display only supports 1-bit, for example.

> 1. The user really doesn't know the color depth in advance?

He might not know the color depth **before building the firmware image**. He must know the color depth when attaching the screen and selecting the right display driver.

> 2\. Is it really a use-case to have more large displays simultaneously with different color depth? (when the color depth conversation takes too much time)

I don't really know. I imagine not too many users really need multi-display with different display resolutions.

embeddedt, 2019-03-03:
> > 3. Now the images are generated will all possible color depths and a define selects the corresponding format based on LV_COLOR_DEPTH. With run-time color depth config, we would lose this simple and convenient behavior.

> Not necessarily.
> You could use function pointers to configure on runtime which function is called (different function for each color depth).

It's not just function pointers; it's also the bitmap data itself. Have a look at this: https://raw.githubusercontent.com/littlevgl/lv_examples/master/lv_apps/demo/img_bubble_pattern.c

Supporting dynamic color depths would require us to embed all possible formats into the image. This would cause a significant increase in Flash usage.

amirgon, 2019-03-03:
> It's not just function pointers; it's also the bitmap data itself. Have a look at this: https://raw.githubusercontent.com/littlevgl/lv_examples/master/lv_apps/demo/img_bubble_pattern.c
>
> Supporting [...]

@embeddedt I'm not sure I understand. Your link points to `lv_examples` repo, not to lvgl. I would expect images to be kept as files on the flash and not as compiled programs.  
With files, the user can select **on runtime** what format and resolution he would like to use. So for images only one format would be needed not all possible formats as you suggest.

embeddedt, 2019-03-03:
@amirgon 
> I would expect images to be kept as files on the flash and not as compiled programs.

It's important to remember that this feature applies to C as well as Micropython. Many users are likely to use the image generator to generate images, not store them on a filesystem.

> With files, the user can select on runtime what format and resolution he would like to use.

Correct. With files, this isn't a problem.

amirgon, 2019-03-03:
> It's important to remember that this feature applies to C as well as Micropython. Many users are likely to use the image generator to generate images, not store them on a filesystem.

@embeddedt That's fine with me. In case of generating images as C code, you can still rely on color depth macros.  
My assumption is that micropython users will not use the image generator, but will use image files instead, since the point in micropython is not to compile C code.

embeddedt, 2019-03-03:
@amirgon How should we handle the case where a generated C image contains a color depth that is different from the display?

amirgon, 2019-03-03:
> How should we handle the case where a generated C image contains a color depth that is different from the display?

@embeddedt Only images with format that matches the current color configuration can be displayed.  
If this is an issue, we can let the users select between **compile time** color depth and **runtime** color depth configuration.  

1. If the user defined color depth macros, it would affect both the image generator and lvgl core, and will prevent configuring image depth on runtime.  
2. But if the user did not define color depth macros, he will be able to configure it on runtime (but not use C generated images if they don't match his runtime configuration).   

I would expect micropython users to only use the second option (runtime configuration). Pure-C users can select the first option if the want, especially if they need the image generator.

kisvegabor, 2019-03-04:
**Preprogrammed MCU **
I see your point and we really need to take into account this scenario.

**Image handling**
With files, it really can work. The output format [convert](https://littlevgl.com/image-to-c-array) can be C array or Binary with a specific color format. So it works this way already. 

**Run time color depth in general**
> That's right. It takes more cycles, but it also means more RAM consumption, right? If color is configured as 32bit but the display only supports 1-bit, for example.

Let's say you have a large 800x480 display and having 1/20 screen sized VDB. 
With 32 bit color depth it needs `800 x 480 x 4 / 20 = 75 kB` On 16 bit the half of it, `37.5 kB`. So you really lose "only" 37.5 kB. However, with such a big display and Micropython support, you should have a few extra RAM.

1 bpp displays are usually small. Assuming 1/8 screen sized VDB:  `128 x 64 x 4 / 8 = 4kB`. On 1bpp color depth, it needs 1/4 memory: 1 kB.  So that 3 kB doesn't really matter.

amirgon, 2019-03-04:
@kisvegabor So summarizing the color-depth discussion above:

1. You say that changing lvgl to configure color-depth on runtime is difficult and comes at a cost.
2. We can configure micropython to always use 32-bit color-depth.  
In case of a display with a different color depth, the display driver will need to do the conversion on the `disp_flush`  callback.
3. Having this conversion on the display driver will cost a little more cycles and RAM, but MCUs that are capable enough to run micropython would probably have enough resources to handle it.
4. Display resolution and orientation will be configurable on runtime.  

If so, I understand the limitation and I hope we can live with it.

kisvegabor, 2019-03-04:
> So summarizing the color-depth discussion above

Yes, for all points :)

> But what about other parameters?
- `LV_VDB_SIZE`, `LV_VDB_DOUBLE`, `LV_VDB_TRUE_DOUBLE_BUFFERED` are already run-time configurable. See: https://github.com/littlevgl/lvgl/blob/event/lv_porting/lv_port_disp_templ.c#L56
- `LV_REFR_PERIOD`: it's a global parameter and not related to a specific display. It means how often redraw the dirty areas. Do you think it should be set (or overwritten) in runtime for each display?
- `LV_INDEV_*` some of them can be run time configures. Still need to be examined.
- `LV_COLOR_SCREEN_TRANSP, LV_COLOR_TRANSP` no problem with them I can move them to the driver.

embeddedt, 2019-03-04:
> `LV_REFR_PERIOD`: it's a global parameter and not related to a specific display. It means how often redraw the dirty areas. Do you think it should be set (or overwritten) in runtime for each display?

It would be convenient, i.e. for a platform that had an eInk display and a regular TFT display you would want different refresh rates. I don't think it's mandatory, however, so if it complicates the implementation it's not worth it.

amirgon, 2019-03-04:
> * `LV_REFR_PERIOD`: it's a global parameter and not related to a specific display. It means how often redraw the dirty areas. Do you think it should be set (or overwritten) in runtime for each display?

@kisvegabor Even if not for each display, it could be useful to set it up on runtime to avoid the need to rebuild firmware image in order to change it.

kisvegabor, 2019-03-05:
Now one task integrates through all displays and refreshes them. However, it would be possible to have a unique `refr_task` for each display. It's easy to implement and would give the highest flexibility.

kisvegabor, 2019-03-30:
I made the following updates:
- `antialiasing` is moved to the display driver. `#define LV_ANTIALIASING  1` is required to globally enable anti-aliasing but in can be enabled/disabled for each display.
- Each display has its own `refr_task` which can be get with `lv_disp_get_refr_task`. The returned `lv_task_t` can be modified with `lv_task_...` function
- `rotate` is added to driver to swap horizontal and vertical resolution
- `lv_disp_update_drv` is added update the display's driver in run time. If the resolution or rotation is changed the screens will be resized accordingly.

kisvegabor, 2019-03-31:
@amirgon does it fit well into the Micropython binding?

@fpw you asked me to notify you when run time resolution change is implemented.

amirgon, 2019-03-31:
> @amirgon does it fit well into the Micropython binding?

@kisvegabor I'm in the process of making some changes in micropython binding for `dev-6.0`.  
If I see any issues with your updates, I'll let you know.

kisvegabor, 2019-04-04:
I made the input devices run-time configurable as well. Form now in `lv_conf.h` you can set the default values. 

Thank you very much for your contribution!

If you find some bugs please open a new issue for them.

closed 2019-04-04 by kisvegabor