lvgl/lvgl #332: lv_label_set_text memory allocation issue.

nameofuser1, 2018-07-24:
Here are lines 156 and 157 of `lv_label.c`:

    ext->text = lv_mem_alloc(len);
    strcpy(ext->text, text);

As you can see if memory is not allocated there is no check for NULL and therefore `strcpy` will likely cause a crash like it is in my case.

kisvegabor, 2018-07-25:
Hi,

I pushed an update to `dev-5.2` branch to handle memory allocation failures in the library.
See: https://github.com/littlevgl/lvgl/blob/dev-5.2/lv_objx/lv_label.c#L161

I will extend the `lv_mem_assert` macro to add log if the memory allocation is failed.

nameofuser1, 2018-07-25:
Thanks! That will do.

Also please check these `lv_mem_realloc` outputs the also miss memory validation:

https://github.com/littlevgl/lvgl/blob/2e17562e51e5b0a058abbd0be19dc10d1d1b5cf8/lv_objx/lv_label.c#L114

https://github.com/littlevgl/lvgl/blob/2e17562e51e5b0a058abbd0be19dc10d1d1b5cf8/lv_objx/lv_label.c#L150

https://github.com/littlevgl/lvgl/blob/2e17562e51e5b0a058abbd0be19dc10d1d1b5cf8/lv_objx/lv_label.c#L593

kisvegabor, 2018-07-25:
Thanks, I forget the realloc-s. I pushed the update.

closed 2018-07-25 by nameofuser1