lvgl/lvgl #335: Button style is implicitly updated when actions occurs.

nameofuser1, 2018-07-26:
It looks like buttons are redrawn on the screen when action occurs regardless of their explicit modifications. I understand that this behavior allows to update styles without handling events explicitly. However I think it would be better if there was a feature that allows to explicitly specify whether I would like to change the style on event or not. This can be justified by the fact that using slow LCD screens and updating UI elements without any changes in their style looks quite awful.

As a solution I would propose to create additional field, something like `update_style_on_event`, and functions which will set and clear this flag. In signal processing function check for this flag and redraw only if needed.

kisvegabor, 2018-07-27:
Hi,

As you said the button is redrawn because a new style is set for it on press/release. The `lv_obj_style()` function will invalidate the button. But if we added a line to `lv_obj_set_style` to invalidate the object only if the new style is different from the current the problem would be solved by setting the same style in pressed and released state.

What do you think?

nameofuser1, 2018-07-30:
Didn't notice that styles are saved as pointers. Yes, I agree this would be a better solution.

kisvegabor, 2018-08-01:
Great!

Then please add `if(obj->style_p == style) return;` to the very beginning of `lv_obj_set_style` and let's see if it's help.

kisvegabor, 2018-08-09:
Implemented in #352 Pull request.

Thank you!

closed 2018-08-09 by kisvegabor