lvgl/lvgl #336: How to speed up drawing

JefkeB, 2018-07-27:
Hi,

Not really an issue but more a question.

I'm working on a port to get the library running on a LPCXpresso54608 board.
It is working so far but is kinda slow.
Especially when 'sliding' from 1 page to another in the demo.
There is no dma2d like in the STM32 parts  available.
What are the cheapest (memory\ram) tricks\solutions to get it up to speed ?

Jef

kisvegabor, 2018-07-28:
Hi,

I see that you have powerful MCU. But what is your screen resolution, and how is it connected to the MCU? SPI, parallel port or the MCU's LCD interface with internal or external frame buffer?

Anyway, basically, you can do these steps:
- The most simple thing is to turn on compiler optimization.
- If you have RAM you can increase VDB size (`LV_VDB_SIZE` in lv_conf.h)
- If you have more RAM and can do the `disp_flush` in the background (DMA?) you can enable `LV_CDB_DOUBLE` (lv_conf.h). This way the rendering and refreshing of the display will be executed parallelly.

JefkeB, 2018-07-28:
Thanks for your tips.

Indeed a lot of power Cortex-M4 running @ 180Mhz with a dedicated LCD controller and 16MB of external sdram. The resolution is 480x272 with a capacitive touchscreen.

As there is (sd)ram enough I did the following
- Put the VDB buffer in sdram and made it the full size of the screen.
This was already a big improvement, still some tearing but much better.
- After this I implemented a rough test version with dma, this improves it even more. Almost all of the tearing is gone. When moving the finger on the screen very fast between two pages (list & chart) there is sometimes some tearing visible.

So for now it seems to be working perfect

Next step will be porting the library to the Renesas RX65N platform.
I'm in the process of comparing some platforms (also the STM32F7 discovery).
Just for the fun of it, these are all board 'collected' during some workshops and are waiting for a project and LittlevGL seems a nice library for using as a gui on them.

Thanks again,

Jef.

closed 2018-07-28 by JefkeB

kisvegabor, 2018-07-29:
Hi,

Great! :)

Putting VDB to SDRAM can be slow because LittlevGL reads/writes the VDB very often. E.g. to render one pixel the library might write that pixel 3 times: background + button + label. And accessing the SDRAM is slower than accessing the internal RAM. 

So placing the framebuffer into the external seems a better choice because the frame buffer's pixels are written only once when `disp_flush` is called.