lvgl/lvgl #337: last tested version?

Masi2017, 2018-07-27:
Hi all,

I was wondering if I can download v5.2 and issue #207 has been resolved in this version? since I tested beta version and it was fine but don't know about v5.2.

Thanks in advance

kisvegabor, 2018-07-28:
Hi,

Yes, I already merged this fix into dev-5.2.
But if you try it, please verify that it works there too.

Masi2017, 2018-08-15:
I tested here as well and it works fine the only issue I have is , assume that there are 4 objects obj 3 and 4 have same parent, assume that the object 4 has been focused, if I  delete the parent of object 3 and 4 ,  I expected that object 2 which is the previous object be focused, but object 1 is focused while if object 3 is focused and parent is deleted, object 2 is selected. I would appreciate if you let me know if this is an issue or it is as expected and implemented?

kisvegabor, 2018-08-15:
I tested it with this code and for me the first object became focused in both cases. When the focused object is deleted the **next** object is focused in the current implementation. 
```c
  /*Create a group*/
    lv_group_t * g = lv_group_create();

    /*Create a Button 1 on the screen*/
    lv_obj_t * label;
    lv_obj_t * btn1 = lv_btn_create(lv_scr_act(), NULL);
    lv_group_add_obj(g, btn1);
    lv_obj_set_pos(btn1, 10, 10);
    label = lv_label_create(btn1, NULL);
    lv_label_set_text(label, "1");

    /*Create a Button 2 on the screen*/
    lv_obj_t * btn2 = lv_btn_create(lv_scr_act(), NULL);
    lv_group_add_obj(g, btn2);
    lv_obj_set_pos(btn2, 10, 100);
    label = lv_label_create(btn2, NULL);
    lv_label_set_text(label, "2");

    /*Create a parent for Button 3 and 4*/
    lv_obj_t * par = lv_obj_create(lv_scr_act(), NULL);
    lv_obj_set_size(par, 200, 200);
    lv_obj_set_pos(par, 200, 0);

    /*Create a Button 3 on the par*/
    lv_obj_t * btn3 = lv_btn_create(par, NULL);
    lv_group_add_obj(g, btn3);
    lv_obj_set_pos(btn3, 10, 10);
    label = lv_label_create(btn3, NULL);
    lv_label_set_text(label, "3");

    /*Create a Button 4 on the par*/
    lv_obj_t * btn4 = lv_btn_create(par, NULL);
    lv_group_add_obj(g, btn4);
    lv_obj_set_pos(btn4, 10, 100);
    label = lv_label_create(btn4, NULL);
    lv_label_set_text(label, "4");

    /*TEST HERE: Focus on btn3 or btn4*/
    lv_group_focus_obj(btn3);
    lv_obj_del(par);
```

Masi2017, 2018-08-16:
Yes your right, my test was different. In fact after pressing the Prev key, I tried to delete the parent in focus group cb function ( to me is like going back to the previous menu) . If first object of parent is selected, by pressing  Prev , previous object will be selected (for example btn2 in your sample code) and then parent is deleted but if second object of parent is selected, by pressing Prev , first object is selected (btn3 in your sample code)  and then once parent is deleted next object is selected (btn1 in your sample code). This is the reason why I've seen the issue. I have to do something to deal with my issue since I created three menus , and in the third menu I have 6 btns and once user presses the Prev key (no matter which btn from the third menu is focused ) the previous menu should be focused. However, library works as promising.  
Thank you!

kisvegabor, 2018-08-17:
I see. Maybe you should store a pointer to the first button in each menu and directly call `lv_group_focus_obj` when pressing  "PREV".

As it the issue seems solved I close it.

As always feel free to reopen it, if is doesn't work correctly.

closed 2018-08-17 by kisvegabor