lvgl/lvgl #338: Memory leak with "lv_obj_set_hidden"

Zaltora, 2018-07-28:
The code that show me the bug:
```c
    if(_xUartEvent.ucRx)
    {
        lv_obj_set_hidden(pxPrvArgs->apxObjects[1], false);
        _xUartEvent.ucRx = 0;
    }
    else
    {
        lv_obj_set_hidden(pxPrvArgs->apxObjects[1], true);
    }
    if(_xUartEvent.ucTx)
    {
        lv_obj_set_hidden(pxPrvArgs->apxObjects[2], false);
        _xUartEvent.ucTx = 0;
    }
    else
    {
        lv_obj_set_hidden(pxPrvArgs->apxObjects[2], true);
    }
    return 0;
```
Objects are displayed and disable repetitively.  The heap decrease constantly.

I didn't use VDB, i use custom memory allocation "malloc" and "free".
Extra "malloc" or missing "free" somewhere.

kisvegabor, 2018-07-29:
Hi,

It strange because there shouldn't be memory allocation because of `lv_obj_set_hidden`.
 
I tested it with `dev-5.2` like this:
```c
    lv_obj_t * obj = lv_obj_create(lv_scr_act(), NULL);
    lv_mem_monitor_t mon;
    uint32_t i = 0;

    while(1) {
        /* Periodically call the lv_task handler.
         * It could be done in a timer interrupt or an OS task too.*/
        lv_task_handler();
        usleep(10000);       /*Just to let the system breathe */
        lv_obj_set_hidden(obj, i % 2 ? true : false);
        lv_mem_monitor(&mon);
        printf("after: free %d, frag %d\n", mon.free_size, mon.frag_pct);
        i++;
    }
```

And the result:
```
after: free 31488, frag 0
after: free 31488, frag 0
after: free 31488, frag 0
after: free 31488, frag 0
after: free 31488, frag 0
after: free 31488, frag 0
after: free 31488, frag 0
after: free 31488, frag 0
after: free 31488, frag 0
after: free 31488, frag 0
after: free 31488, frag 0
after: free 31488, frag 0
...
```

So It's fine for me. please check these:
1. Is it really fine if you not call `lv_obj_set_hidden`?
2. What if you continuously change the style instead of making the object hidden?

Zaltora, 2018-07-30:
i confirm you that: 
1) If i comment the lines above in my program, the memory leak disappear ! (only use hidden API)
2) I got same memory leak problem with style API

```c
static int _iScreenUsbUpdate(CoreArgs_t* pxPrvArgs, void* pvArgs)
{
    //Uart_s* pxUartvents = (Uart_s*)pvArgs;
    if(_xUartEvent.ucRx)
    {
        lv_obj_set_style(pxPrvArgs->apxObjects[1], &style_shadow);
        _xUartEvent.ucRx = 0;
    }
    else
    {
        lv_obj_set_style(pxPrvArgs->apxObjects[1], &style_line3);
    }
    if(_xUartEvent.ucTx)
    {
        lv_obj_set_style(pxPrvArgs->apxObjects[2], &style_shadow);
        _xUartEvent.ucTx = 0;
    }
    else
    {
        lv_obj_set_style(pxPrvArgs->apxObjects[2], &style_line3);
    }
    return 0;
}
```
I got 3 function in my program by windows. One function called once, one for dynamic change called each loop, and one to manage button (custom implementation).

The details of objects created (once):
```c
    static lv_style_t style_line3;
    lv_style_copy(&style_line3, &lv_style_plain);
    style_line3.line.color = LV_COLOR_MAKE(0x00, 0x3b, 0x75);
    style_line3.line.width = 5;

    /*Create line with default style*/
    pxPrvArgs->apxObjects[1] = lv_line_create(lv_scr_act(), NULL);
    lv_line_set_points(pxPrvArgs->apxObjects[1], line_points, 2);     /*Set the points*/
    lv_obj_align(pxPrvArgs->apxObjects[1], pxPrvArgs->apxObjects[0], LV_ALIGN_OUT_RIGHT_TOP, 0, 0);
    lv_line_set_style(pxPrvArgs->apxObjects[1], &style_line3);

    pxPrvArgs->apxObjects[2] = lv_line_create(lv_scr_act(), NULL);
    lv_line_set_points(pxPrvArgs->apxObjects[2], line_points, 2);     /*Set the points*/
    lv_obj_align(pxPrvArgs->apxObjects[2], pxPrvArgs->apxObjects[0], LV_ALIGN_OUT_RIGHT_BOTTOM, 0, 0);
    lv_line_set_style(pxPrvArgs->apxObjects[2], &style_line3);
```

My config file:  [lv_conf.txt](https://github.com/littlevgl/lvgl/files/2240096/lv_conf.txt)

kisvegabor, 2018-08-01:
And what happens if you use the built-in memory manager functions (`LV_MEM_CUSTOM     0`) and monitor the memory usage like this:

```c
lv_mem_monitor_t mon;
lv_mem_monitor(&mon);
printf("after: free %d, frag %d\n", mon.free_size, mon.frag_pct);
```

By the way, how do you know the used heap size using malloc/free?

Zaltora, 2018-08-01:
Yes, i use FreeRTOS and i got a function to get total heap: `xPortGetFreeHeapSize()`
Test that will follow use "hidden API code at the first comment"

Test 1:
Setting:
```
#define LV_MEM_CUSTOM      0
#define LV_MEM_SIZE    (8U * 1024U)  
```
Output:
```
<CMD>   rtos heap
<CMD>   Heap: 5280
<CMD>   lvgl: free 7220, frag 0

<CMD>   rtos heap
<CMD>   Heap: 5272
<CMD>   lvgl: free 7140, frag 0

<CMD>   rtos heap
<CMD>   Heap: 5272
<CMD>   lvgl: free 7068, frag 0

<CMD>   rtos heap
<CMD>   Heap: 5272
<CMD>   lvgl: free 6996, frag 0

<CMD>   rtos heap
<CMD>   Heap: 5272
<CMD>   lvgl: free 6916, frag 0
```
Test 2:
Setting:
```
#define LV_MEM_CUSTOM      1
```
Output:
```
<CMD>   rtos heap
<CMD>   Heap: 12276
<CMD>   lvgl: free 0, frag 0

<CMD>   rtos heap
<CMD>   Heap: 12164
<CMD>   lvgl: free 0, frag 0

<CMD>   rtos heap
<CMD>   Heap: 12036
<CMD>   lvgl: free 0, frag 0

<CMD>   rtos heap
<CMD>   Heap: 11876
<CMD>   lvgl: free 0, frag 0
```

Test 3 with the code commented:
Setting:
```
#define LV_MEM_CUSTOM      0
#define LV_MEM_SIZE    (8U * 1024U)  
```
Output:
```
<CMD>   rtos heap
<CMD>   Heap: 5280
<CMD>   lvgl: free 7252, frag 0

<CMD>   rtos heap
<CMD>   Heap: 5264
<CMD>   lvgl: free 7252, frag 0

<CMD>   rtos heap
<CMD>   Heap: 5272
<CMD>   lvgl: free 7252, frag 0

<CMD>   rtos heap
<CMD>   Heap: 5264
<CMD>   lvgl: free 7252, frag 0

```

Zaltora, 2018-08-01:
Each time i use UART to write the command "rtos heap" or each time the system write into "UART", the hidden API is called and i can see in the screen device transmission indicator.
Unfortunately, calls to the "hidden" and "style" system do not free up all memory. (Apparently)

kisvegabor, 2018-08-01:
In the PC simulator, I checked that no memory manager function is called due to `lv_obj_set_hidden()`.

I suggest to start with a very simple test:
```c
    lv_obj_t * obj = lv_obj_create(lv_scr_act(), NULL);
    lv_mem_monitor_t mon;

    uint32_t i = 0;
    for(i = 0; i <= 10; i++) {
    	lv_obj_set_hidden(obj, i % 2 ? true : false);
	lv_mem_monitor(&mon);
	printf("after: free %d, frag %d\n", mon.free_size, mon.frag_pct);
    }
```

For me, it not shows memory leak. For you?

Zaltora, 2018-08-02:
No memory leak problem with your code, but if i change one line, i got the problem. It is the "line" object that got the problem.
Test this code below:
 ```c
    lv_obj_t * obj = lv_line_create(lv_scr_act(), NULL);
    lv_mem_monitor_t mon;

    uint32_t i = 0;
    for(i = 0; i <= 10; i++)
    {
        lv_obj_set_hidden(obj, i % 2 ? true : false);
        lv_mem_monitor(&mon);
        printf("after: free %d, frag %d\n", mon.free_size, mon.frag_pct);
    }
```

kisvegabor, 2018-08-09:
With your code I got this:
```c
after: free 30976, frag 0
after: free 30976, frag 0
after: free 30976, frag 0
after: free 30976, frag 0
after: free 30976, frag 0
after: free 30976, frag 0
after: free 30976, frag 0
after: free 30976, frag 0
after: free 30976, frag 0
after: free 30976, frag 0
after: free 30976, frag 0
```

With `printf`s or a debugger can you figure out which memory manager function is called from `lv_mem.c` durint lv_obj_set_hidden? (lv_mem_alloc, lv_mem_realloc)

Zaltora, 2018-08-10:
After doing a clean installation:
- reinstall lvgl
- reinstall lv_drivers
- update lv_conf.h (instead of backport him)
- regenerate my fonts/symbol (for ASCII usage)

The problem disappear... i can't tell you what was wrong now... i am not sure if it is a good new but anyway it is work.

kisvegabor, 2018-08-10:
It's strange. The important thing is, it's working well with the latest version.

closed 2018-08-21 by kisvegabor