lvgl/lvgl #339: [5.2-dev]  Missing preprocessor inclusion

Zaltora, 2018-07-28:
My setting:
#define LV_VDB_SIZE         0 
#define USE_LV_REAL_DRAW        1
#define USE_LV_IMG      1

file : `lv_draw.c`
```c
#if LV_VDB_SIZE != 0
void (*px_fp)(lv_coord_t x, lv_coord_t y, const lv_area_t * mask, lv_color_t color, lv_opa_t opa) = lv_vpx;
void (*fill_fp)(const lv_area_t * coords, const lv_area_t * mask, lv_color_t color, lv_opa_t opa) =  lv_vfill;
void (*letter_fp)(const lv_point_t * pos_p, const lv_area_t * mask, const lv_font_t * font_p, uint32_t letter, lv_color_t color, lv_opa_t opa) = lv_vletter;
#  if USE_LV_IMG
void (*map_fp)(const lv_area_t * cords_p, const lv_area_t * mask_p,
               const uint8_t * map_p, lv_opa_t opa, bool chroma_key, bool alpha_byte,
               lv_color_t recolor, lv_opa_t recolor_opa) = lv_vmap;
#  endif /*USE_LV_IMG*/
#elif USE_LV_REAL_DRAW != 0
/* px_fp used only by shadow drawing and anti aliasing */
void (*px_fp)(lv_coord_t x, lv_coord_t y, const lv_area_t * mask, lv_color_t color, lv_opa_t opa) = lv_rpx;
void (*fill_fp)(const lv_area_t * coords, const lv_area_t * mask, lv_color_t color, lv_opa_t opa) =  lv_rfill;
void (*letter_fp)(const lv_point_t * pos_p, const lv_area_t * mask, const lv_font_t * font_p, uint32_t letter, lv_color_t color, lv_opa_t opa) = lv_rletter;
#  if USE_LV_IMG
void (*map_fp)(const lv_area_t * cords_p, const lv_area_t * mask_p,
                      const uint8_t * map_p, lv_opa_t opa, bool chroma_key, bool alpha_byte,
                      lv_color_t recolor, lv_opa_t recolor_opa) = lv_rmap;
#  endif /*USE_LV_IMG*/
#else
/*Invalid settings. Compiler error will be thrown*/
void (*px_fp)(lv_coord_t x, lv_coord_t y, const lv_area_t * mask, lv_color_t color, lv_opa_t opa) = NULL;
void (*fill_fp)(const lv_area_t * coords, const lv_area_t * mask, lv_color_t color, lv_opa_t opa) =  NULL;
void (*letter_fp)(const lv_point_t * pos_p, const lv_area_t * mask, const lv_font_t * font_p, uint32_t letter, lv_color_t color, lv_opa_t opa) = NULL;
void (*map_fp)(const lv_area_t * cords_p, const lv_area_t * mask_p,
                      const uint8_t * map_p, lv_opa_t opa, bool chroma_key, bool alpha_byte,
                      lv_color_t recolor, lv_opa_t recolor_opa) = NULL;
#endif
```
file:` lv_draw.h`

```c
extern void (*px_fp)(lv_coord_t x, lv_coord_t y, const lv_area_t * mask, lv_color_t color, lv_opa_t opa);
extern void (*fill_fp)(const lv_area_t * coords, const lv_area_t * mask, lv_color_t color, lv_opa_t opa);
extern void (*letter_fp)(const lv_point_t * pos_p, const lv_area_t * mask, const lv_font_t * font_p, uint32_t letter, lv_color_t color, lv_opa_t opa);
extern void (*map_fp)(const lv_area_t * cords_p, const lv_area_t * mask_p,
               const uint8_t * map_p, lv_opa_t opa, bool chroma_key, bool alpha_byte,
               lv_color_t recolor, lv_opa_t recolor_opa);
```

The problem with my setting: the compiler don't found `px_fp` declaration. The reason, it is called even if `px_fp` is used only by shadow drawing and anti aliasing.

To fix my problem, i just remove comment and declare it. I don't no if it i was the best way ?
If it is not used like the comment said, it is better to add new preprocessor inclusion to remove theses call when theses settings are set. What it is the best  ?
Need patch this file : `lv_draw_line.c`
I will do a pull request once i got the answer :)

PS: preprocessor inclusion are sources of recurring errors. I observe that when lvgl will only work with a VDB, a lot of configuration like these will be removed. It is a good thing.

kisvegabor, 2018-07-29:
Hi,

Thank you for pointing this out. In `dev-5.2` I moved the draw algorithms to separate files. I didn't try the real draw configuration so far.

I pushed an update to fix it. I made it clearer:

```c
#if LV_VDB_SIZE != 0
void (*px_fp)(lv_coord_t x, lv_coord_t y, const lv_area_t * mask, lv_color_t color, lv_opa_t opa) = lv_vpx;
void (*fill_fp)(const lv_area_t * coords, const lv_area_t * mask, lv_color_t color, lv_opa_t opa) =  lv_vfill;
void (*letter_fp)(const lv_point_t * pos_p, const lv_area_t * mask, const lv_font_t * font_p, uint32_t letter, lv_color_t color, lv_opa_t opa) = lv_vletter;
void (*map_fp)(const lv_area_t * cords_p, const lv_area_t * mask_p,
               const uint8_t * map_p, lv_opa_t opa, bool chroma_key, bool alpha_byte,
               lv_color_t recolor, lv_opa_t recolor_opa) = lv_vmap;
#else
void (*px_fp)(lv_coord_t x, lv_coord_t y, const lv_area_t * mask, lv_color_t color, lv_opa_t opa) = lv_rpx;
void (*fill_fp)(const lv_area_t * coords, const lv_area_t * mask, lv_color_t color, lv_opa_t opa) =  lv_rfill;
void (*letter_fp)(const lv_point_t * pos_p, const lv_area_t * mask, const lv_font_t * font_p, uint32_t letter, lv_color_t color, lv_opa_t opa) = lv_rletter;
void (*map_fp)(const lv_area_t * cords_p, const lv_area_t * mask_p,
                      const uint8_t * map_p, lv_opa_t opa, bool chroma_key, bool alpha_byte,
                      lv_color_t recolor, lv_opa_t recolor_opa) = lv_rmap;
#endif
``` 

As these `.._fp` functions are used or will be used more widely it is not reasonable to ignore them by defines. E.g.:
- `px_fp` is used not only by shadow but used by lv_line too.
- `map_fp` will be used by a new object (image button) and not only by `lv_img`.

So to test all options seems quite complicated to me. In addition, compilers can ignore unused functions. E.g. with gcc: https://stackoverflow.com/questions/6687630/how-to-remove-unused-c-c-symbols-with-gcc-and-ld

What do yo think?

Zaltora, 2018-07-30:
It is better to limit the use of the preprocessor. The code will be easier to maintain and understand.
i agree with your changes.

Just one question, the symbol describe in this file is constant ? add the keyword  `const` is good ?

Zaltora, 2018-07-31:
I got some warning to fix too for dev-5.2:
line 232: `lv_area_t circle_area;` lv_objx/lv_line.c   (set but not used)
line 186: `lv_coord_t width_safe = width;` lv_draw/lv_draw_line.c (not used)
line 111: `lv_rletter_set_background` lv_draw/lv_draw_label.c (implicit declaration)
line 1402: `antialias_get_opa_circ` lv_draw/lv_draw_rect.c (defined but not used)

kisvegabor, 2018-08-01:
Yes, it is constant. Good idea to add the `const` prefix.

Some parts of `dev-5.2` are still work-in-progress and it causes some warnings. They will be fixed soon. Thank you!

Zaltora, 2018-09-21:
Updated to last commit. Compilation error.
```
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_core/lv_refr.c: In function 'lv_refr_area_no_vdb':
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_core/lv_refr.c:287:47: error: 'start_mask' undeclared (first use in this function)
     lv_refr_obj_and_children(lv_layer_top(), &start_mask);
```
I will try to found why.

Zaltora, 2018-09-21:
Feel to miss some code when VDB is no used. Can you verify it ? The declaration of the variable is missing and maybe other stuff.

Zaltora, 2018-09-21:
After fixing the problem ( i guess), the build pass but with a lot of warning:

```
CC /opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_line.c
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_line.c: In function 'line_draw_hor':
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_line.c:158:12: warning: function with qualified void return type called [enabled by default]
     fill_fp(&draw_area, mask, style->line.color, opa);
            ^
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_line.c: In function 'line_draw_ver':
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_line.c:180:12: warning: function with qualified void return type called [enabled by default]
     fill_fp(&draw_area, mask, style->line.color, opa);
            ^
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_line.c: In function 'line_draw_skew':
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_line.c:389:24: warning: function with qualified void return type called [enabled by default]
                 fill_fp(&draw_area, mask, style->line.color, opa);
                        ^
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_line.c:394:26: warning: function with qualified void return type called [enabled by default]
                     px_fp(draw_area.x1 , draw_area.y1 - main_line->sy, mask, style->line.color, opa);
                          ^
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_line.c:416:20: warning: function with qualified void return type called [enabled by default]
             fill_fp(&draw_area, mask, style->line.color , opa);
                    ^
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_line.c:421:22: warning: function with qualified void return type called [enabled by default]
                 px_fp(draw_area.x1, draw_area.y1 - main_line->sy, mask, style->line.color, opa);
                      ^
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_line.c:442:24: warning: function with qualified void return type called [enabled by default]
                 fill_fp(&draw_area, mask, style->line.color, opa);
                        ^
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_line.c:447:26: warning: function with qualified void return type called [enabled by default]
                     px_fp(draw_area.x1 - main_line->sx, draw_area.y1, mask, style->line.color, opa);
                          ^
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_line.c:472:20: warning: function with qualified void return type called [enabled by default]
             fill_fp(&draw_area, mask, style->line.color, opa);
                    ^
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_line.c:477:22: warning: function with qualified void return type called [enabled by default]
                 px_fp(draw_area.x1 - main_line->sx, draw_area.y1, mask, style->line.color, opa);
                      ^
CC /opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_vbasic.c
CC /opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_label.c
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_label.c: In function 'lv_draw_label':
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_label.c:112:5: warning: implicit declaration of function 'lv_rletter_set_background' [-Wimplicit-function-declaration]
     lv_rletter_set_background(style->body.main_color);
     ^
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_label.c:167:22: warning: function with qualified void return type called [enabled by default]
             letter_fp(&pos, mask, font, letter, color, opa);
                      ^
CC /opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_rect.c
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_rect.c: In function 'lv_draw_rect_main_mid':
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_rect.c:136:16: warning: function with qualified void return type called [enabled by default]
         fill_fp(&work_area, mask, mcolor, opa);
                ^
```
```
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_rect.c:160:20: warning: function with qualified void return type called [enabled by default]
             fill_fp(&work_area, mask, act_color, opa);
                    ^
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_rect.c: In function 'lv_draw_rect_main_corner':
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_rect.c:307:20: warning: function with qualified void return type called [enabled by default]
             fill_fp(&edge_top_area, mask, act_color, opa);
                    ^
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_rect.c:316:20: warning: function with qualified void return type called [enabled by default]
             fill_fp(&mid_top_area, mask, act_color, opa);
                    ^
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_rect.c:325:20: warning: function with qualified void return type called [enabled by default]
             fill_fp(&mid_bot_area, mask, act_color, opa);
                    ^
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_rect.c:335:20: warning: function with qualified void return type called [enabled by default]
             fill_fp(&edge_bot_area, mask, act_color, opa);
                    ^
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_rect.c:367:12: warning: function with qualified void return type called [enabled by default]
[... 111 lines cut ...]
                ^
CC /opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw.c
[... cut ...]
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_arc.c: In function 'ver_line':
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_arc.c:242:12: warning: function with qualified void return type called [enabled by default]
     fill_fp(&area, mask, color, opa);
            ^
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_arc.c: In function 'hor_line':
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_arc.c:250:12: warning: function with qualified void return type called [enabled by default]
     fill_fp(&area, mask, color, opa);
            ^
```

kisvegabor, 2018-09-23:
Thank you for the feedback!
I pushed an update to solve these warnings.

Zaltora, 2018-09-24:
Close. Stay few warnings:
```
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_img.c:41:1: warning: braces around scalar initializer [enabled by default]
 static lv_color_t decoder_index_map[256] = {{{0}}};
 ^
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_img.c:41:1: warning: (near initialization for 'decoder_index_map[0].blue') [enabled by default]
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_img.c: In function 'lv_img_decoder_open':
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_img.c:411:23: warning: assignment from incompatible pointer type [enabled by default]
             palette_p = (lv_color_t *)((lv_img_dsc_t *)decoder_src)->data;
                       ^
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_img.c: In function 'lv_img_built_in_decoder_line_alpha':
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_img.c:537:16: warning: unused variable 'bg_color' [-Wunused-variable]
     lv_color_t bg_color = decoder_style->image.color;
```

kisvegabor, 2018-09-24:
Thank you! I hope I fixed them :)
I enabled all the warnings but somehow I can't see them.

Zaltora, 2018-09-25:
Got new error + warning:
```
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_img.c: In function 'lv_img_built_in_decoder_line_alpha':
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_img.c:549:2: error: #error "Invalid LV_COLOR_DEPTH. Check it in lv_conf.h"
 #error "Invalid LV_COLOR_DEPTH. Check it in lv_conf.h"
  ^
/opt/esp-dev/esp-open-rtos/lvgl/lvgl/lv_draw/lv_draw_img.c:537:16: warning: unused variable 'bg_color' [-Wunused-variable]
     lv_color_t bg_color = decoder_style->image.color;
                ^
```
My setting: `#define LV_COLOR_DEPTH 1`

---
Strange yes, even interesting:
My gcc version : `xtensa-lx106-elf-gcc (crosstool-NG crosstool-ng-1.22.0-60-g37b07f6f) 4.8.5`
Option i found on it:
```
C_CXX_FLAGS ?= -Wall -Wl,-EL -nostdlib $(EXTRA_C_CXX_FLAGS)
# Flags for C only
CFLAGS		?= $(C_CXX_FLAGS) -std=gnu99 $(EXTRA_CFLAGS)
# Flags for C++ only
CXXFLAGS	?= $(C_CXX_FLAGS) -std=c++0x -fno-exceptions -fno-rtti $(EXTRA_CXXFLAGS)
```

Zaltora, 2018-09-25:
Perfect !

kisvegabor, 2018-09-25:
Great, thank you very much! :+1:

closed 2018-09-25 by kisvegabor