lvgl/lvgl #340: Advanced chart / plotting features

AloyseTech, 2018-07-29:
Hi,

The current chart object is nice but lacks advanced features.

Here is a non-exhaustive list of features that would make lvgl charts perfect for measurement/scientific GUI usage :
- Axis label
- Linear/Log/Time scale type
- Dual axis, left or right (for example to plot voltage an current against time)
- Legend
- Data range selection
- Automatic SI label suffix (ex : base unit 'g', labels are µg mg g kg etc...)
- Line divisor and sub-divisor (for example 1/1 main divisor, 1/10th secondary divisor)

What do you think of these features?

Best regards

kisvegabor, 2018-07-29:
Hi,

Thank you for the suggestion! I agree these are useful and great features. 

I also have two idea: 
- Animation or style change on datapoint selection/click or even show the exact value in a small box.  
- Area chart: similar to the line but the area below the line(s) is filled

AloyseTech, 2018-07-30:
For PC development, I like [QCustomPlot](https://www.qcustomplot.com/). There a a lot of interesting features.

kisvegabor, 2018-08-01:
QCustomPlot looks like a good reference. Firstly a new object (based on lv_chart) should be created which have better features for axis and scales.

Would you like to contribute to this object? If yes, I can help you to get started with creating a new object type and during the whole implementation.

gjohnson0720, 2018-09-26:
I am using a chart for my project and need to add axis labels and tick marks to it.  I would be willing to contribute that work to your project if you walk me through how to do it.

AloyseTech, 2018-09-26:
I personally can’t invest a lot of time right now. But since I will need this for an upcoming project, I should be able to contribute when I start it.

gjohnson0720, 2018-09-26:
@kisvegabor  
Thanks for the primer.  I was not too familiar with your signalling mechanism.  Seems to make sense though.  I tried what you described above, and yes, that works fine.  I also added in tick marks and tick labels inside the graph.

kisvegabor, 2018-09-26:
Awesome! Can you share a screenshot?

gjohnson0720, 2018-09-26:
[image]

kisvegabor, 2018-09-27:
Looks great!
Do you use hardcoded values to test or already use real values of the chart?

gjohnson0720, 2018-09-27:
I'm in the process of adding parameters to the lv_chart_ext_t struct to enable/disable the axis labels and tick marks and also provide the values for the ticks. I need the tick labels to show unit values (e.g., 0 - 10000 meters) instead of data point values (e.g., 0 - 800).  So this will need to be a part of the parameters.

Is that the way that you would recommend to extend the chart control?

kisvegabor, 2018-09-28:
> Is that the way that you would recommend to extend the chart control?

You should place the new parameters in `lv_chart_ext_t` (lv_chart.h) and create `set` `get` API functions for them.

kisvegabor, 2018-10-07:
@gjohnson0720 
Is there any new about it?

gjohnson0720, 2018-10-09:
Yes, I've got something that seems to be working.  I'm happy to share it.  Should I check it in?  If so, to 5.2 branch?

kisvegabor, 2018-10-09:
Great!

v5.2 was released previous week and I've just created the `dev-5.3` branch.
Can make a PR to the `dev-5.3` branch.

gjohnson0720, 2018-10-19:
I have some local commits that i'm trying to push and am getting an error:
remote: Permission to littlevgl/lvgl.git denied to gjohnson0720.
Is this a problem on your end or mine?

kisvegabor, 2018-10-20:
You need to send a Pull request because you don't have write access to the repository:
1. Fork the lvgl repo: Click the Fork button in to top right corner: https://github.com/littlevgl/lvgl
2. lvgl repo will be added to your repositories: https://github.com/gjohnson0720?tab=repositories
3. Clone your new lvgl repo
4. Move your changes there
5. On your repository's main page on GitHub click the  "New pull request" button
6. As base branch set lvgl/dev-5.3
7. Create the pull request
8. See your pull request here: https://github.com/littlevgl/lvgl/pulls

It might seem complicated at first look but it's quite easy once you work on Fork.

More info here: https://help.github.com/articles/creating-a-pull-request/

closed 2018-11-30 by kisvegabor

kisvegabor, 2019-02-06:
There was a [PR](https://github.com/littlevgl/lvgl/pull/472) about adding tick marks but finally, it wasn't merged.    
"area chart" is not supported right now but it can be a new char type. e.g. `LV_CHART_TYPE_AREA`. It can use `lv_draw_triangle` to draw the areas. If you are interested in adding this feature please open a new issue for it.