lvgl/lvgl #341: Will littlevgl support hardware multi layer and 2D features?

abiandy, 2018-07-31:
Dear:

We want to use littlevgl in out SOC platform, which is in the consumer electronics field. 

We have some hardware layers in our display engine, and we have 2D GPU in our SOC, such as scaler, rotage, colorkey, alpha, etc.

So can the littlevgl support hardware layers or the 2D features?

Thanks very much!

kisvegabor, 2018-08-01:
Hi,

LittelvGL has a simple interface to use the GPU. Have look at this:  https://littlevgl.com/porting#display-interface

As you can see now fill and blend is supported but rotate and scale isn't.

kisvegabor, 2018-08-04:
Hi,

we have a discussion about better GPU support in LittlevGL here: https://github.com/littlevgl/lvgl/issues/87

In the [last post](https://github.com/littlevgl/lvgl/issues/87#issuecomment-409696413) I had an idea to treat the layers of the hardware as independent displays. I think it should help to make more out of the GPU/LCD engine.  Unfortunately, the multi-display support will come only in v6.0. (some month later)

However, you can manage to play a video on layer an draw GUI on an other. See: https://github.com/littlevgl/lvgl/issues/262

I hope it helps!

abiandy, 2018-08-04:
Thanks very much for your help! And look forward to the great feature some month later!

closed 2018-08-20 by kisvegabor