lvgl/lvgl #342: text area and action

lidiriel, 2018-07-31:
Hi, is it possible to add click action to an text area ?
I think i can used `lv_page_set_rel_action` because the text area is compose of label and page.

lidiriel, 2018-08-01:
Hi,
 i have modify the object Text Area for adding an action release.
Here a little patch for this.
[0001-add-action-callback-on-text-area.patch.txt](https://github.com/littlevgl/lvgl/files/2249662/0001-add-action-callback-on-text-area.patch.txt)

ps : thanks for this beautiful library.

kisvegabor, 2018-08-01:
Hi,

For your question:
Yes `lv_page_set_rel_action` can be used but it really should have  `lv_ta_set/get_action` functions.

For the patch:
Great, thank you very much! Just one question: Would it be good to not save the action in the Text area? The lv_page saves the action in `lv_page_set_rel_action`. 

So `lv_ta_set_action` can be only a wrapper to `lv_page_set_rel_action` with a `static inline` function in `lv_ta.h`. What do you think?

kisvegabor, 2018-08-20:
I added `lv_ta_set_action` to `dev-5.2`

closed 2018-08-20 by kisvegabor