lvgl/lvgl #344: Gauge object generates unaligned lines for dividers

TMarkivULCRobotics, 2018-07-31:
Couple problems with the gauge that makes it appears to be very imprecise.

1 On the gauge it seems that dividers have different directions instead to be directed to the center of the gauge. 
2 The dividers for labels and lines are not aligned between each other. For example I set the following configuration:
`lv_gauge_set_scale(manometer_obj, angle , 60, 12);` - it means that between each divider with text should be 4 small dividers. But it is not. See attached picture.
[image]

Is that possible to fix? How much effort it needs?

kisvegabor, 2018-08-01:
Hi,

I tested it and it really not works as it should. I will examine it soon.

Thank you for the report.

kisvegabor, 2018-09-04:
Hi, 
I examined the issue. It was because of the bad `line_cnt` value in `lv_gauge_set_scale`.
I added the equation to the API comment about how to calculate `line_cnt` if you know the desired number of subdivision lines: `line_cnt = (sub_div + 1) * (label_cnt - 1) + 1`. So in your case it's 56.

With the direction of the lines, half pixel error always can happen because of rounding. 

[image]

closed 2018-09-04 by kisvegabor