lvgl/lvgl #345: Whether the “ta” object can limit the length of the input text

as564335sa, 2018-08-01:
Whether the “ta” object can limit the length of the input text , i think it is very useful ,and add enhancements that can set the text input type , such as letters or numbers；

kisvegabor, 2018-08-04:
Hi,

I agree both would be useful features. So I added them to `dev-5.2` branch. You can try them with
```c
lv_ta_set_accepted_chars(ta, "0123456789");
lv_ta_set_max_length(ta, 8)
```

as564335sa, 2018-08-07:
Hi,
I update to the `dev-5.2`  branch and test it  , these two functions are good and worked well，but have some issues:
this the code:
```c
lv_obj_t *ta = lv_ta_create(win, NULL);
lv_ta_set_cursor_type(ta, LV_CURSOR_LINE);
lv_ta_set_cursor_pos(ta,LV_TA_CURSOR_LAST);
lv_ta_set_one_line(ta,true);
lv_ta_set_accepted_chars(ta,"0123456789");
lv_ta_set_max_length(ta,5);
lv_ta_set_text(ta,"1\n2aaaaaa");
lv_group_add_obj(g, ta);
```
I got this:
[image]
[image]

you can see:
1、The line is correct when use "enter" key , but is error  when use `lv_ta_set_text`；
2、The max length is correct when input some keys , but is error when use `lv_ta_set_text` ；
3、The letter is error when use `lv_ta_set_text` ; 
3、I del all texts and reinput , it worked well;

kisvegabor, 2018-08-09:
Hi,

I fixed it and now it worked fine for me with every input.

How is it working there?

as564335sa, 2018-08-10:
Great！The issues is solved,thank you!

kisvegabor, 2018-08-10:
Thank you for the test!

closed 2018-08-10 by kisvegabor