lvgl/lvgl #346: Issues about "ta" when "USE_LV_ANIMATION" is set to 0

as564335sa, 2018-08-02:
Hi,
In order to save space and make it run fast，i set  `#define USE_LV_ANIMATION        0`
there are two issues;
1、The cursor will not show，i think it should be show, but not blink；
2、see the code:
```c
passwd_ta = lv_ta_create(cont,NULL);
lv_ta_set_one_line(passwd_ta,true);
lv_obj_align(passwd_ta,cont,LV_ALIGN_CENTER,0,0);
lv_obj_set_size(passwd_ta,200,20);
lv_ta_set_style(passwd_ta,LV_TA_STYLE_BG,&cont_style);
lv_ta_set_pwd_mode(passwd_ta,true);
lv_ta_set_text(passwd_ta,"000000");
```
Because the animation is closed，the "ta" will show "000000" instead of "******" ； If I delete all text and use a keyboard input some text，the first character will not show ‘\*‘；
3、I want show nothing after create the "ta" and don't set up text for a space，if i call `lv_ta_set_text(passwd_ta,NULL);`，it will show the default text；

kisvegabor, 2018-08-04:
Hi,

I fixed both of them in `dev-5.2`. Please test it.

as564335sa, 2018-08-07:
Great! It worked well!

kisvegabor, 2018-08-09:
Thanks for the test! :)

closed 2018-08-09 by kisvegabor