lvgl/lvgl #347: How to integrate with FreeRTOS, etc

markyi370, 2018-08-03:
Hi,

I've discovered LVGL (amazing project!) and I am wondering how to integrate it with an RTOS. The README mentions that OS support is available, and I see that you have tested it with FreeRTOS. However, I don't see any locking mechanisms. Is this required or does it "Just Work (TM)?"

Zaltora, 2018-08-03:
I use lvgl with FreeRTOS. I will do a PR soon on the project `esp-open-rtos` for the esp8266.
Example link with ssd1306 example: [example](https://github.com/Zaltora/esp-open-rtos/tree/lvgl/examples/lvgl_ssd1306)
i use:
lv_drivers => new_api branch
lvgl => dev-5.2 branch

Hope it is help.

DeonMarais64, 2018-08-05:
This is how I do it.

```C
V UI_TakeMutex ( V )
{
	while( xSemaphoreTake( uiMutex, portMAX_DELAY ) != pdTRUE )
		;
}

V UI_GiveMutex ( V )
{
	xSemaphoreGive( uiMutex );
}

static V TaskLVGL( PV p )
{
	for( ; ; )
	{
		vTaskDelay( pdMS_TO_TICKS(10) );

		UI_TakeMutex();
		lv_task_handler();
		UI_GiveMutex();
	}
}

static V TaskUiEvery200ms ( PV p )
{
	UF8 _400ms, _1000ms;

	for( ; ; )
	{
		vTaskDelay( pdMS_TO_TICKS(200) );

		if( ++_400ms >= 2 )
		{
			_400ms = 0;
			Screens_Every400ms();
		}
		if( ++_1000ms >= 5 )
		{
			_1000ms = 0;
			Screens_Every1000ms();
		}
	}
}
```
Screens_Every400ms() and Screens_Every1000ms() call UI_TakeMutex and UI_GiveMutex as and if lvgl API functions are called.

kisvegabor, 2018-08-08:
@markyi370 It seems your question is answered so I close this issue.
If you have further question, feel free to reopen it!

closed 2018-08-08 by kisvegabor

markyi370, 2018-08-09:
@kisvegabor Yes, my question is answered. Thank you for the quick responses everyone!