lvgl/lvgl #348: Does littlevgl support spi flash storage?

relliktei, 2018-08-04:
I am using stm32f429zgt6, how can I store image files into spi flash and load it from spi flash?

kisvegabor, 2018-08-05:
Hi,

LittlevGL has a file system interface however it doesn't have to access to real files. For example, the file name "1" or "2" can refer to the first and second image in the SPI flash.  

You need to write some simple functions (open, close, read etc) to add a driver to your "File system". Read more here: https://github.com/littlevgl/lv_examples/blob/master/lv_tutorial/6_images/lv_tutorial_images.c

turoksama, 2018-08-10:
Need file management? Fatfs is a solution.
Otherwise, I think lvgl is good.

kisvegabor, 2018-08-10:
You don't really need a full-featured filesystem.

For example, if you say this:
- File: "icon1" starts at 0x0000 address
- File: "icon1" starts at 0x1000 address
- File: "icon2" starts at 0x2000 address
- etc

It's very easy to get the address from the file name in your open function.

Then your `read` function should read out the data from the address calculated in `open`. You have to have a read pointer which shows which is the next byte to read.

Still you need a `tell` (to tell the position of your read pointer) and a `seek` ( to set the value of the read pointer) function.

In `open` you will get a pointer to a data with `drv.file_size` size (the size was set when the driver was registered). You can store any data in this pointer, such as the "read pointer", the "base address" etc.

kisvegabor, 2018-09-18:
As there was no activity here for a while I close it to keep the issues clear.

If you have remarks about this topic feel free to comment here and reopen the issue.

ali-rostami, 2019-01-12:
Hi 
how should we define the file_size?

kisvegabor, 2019-01-12:
Hi,

have a look at this: https://github.com/littlevgl/lvgl/issues/609#issuecomment-445963843

kisvegabor, 2019-01-12:
Have a look at https://github.com/littlevgl/lvgl/blob/master/lv_misc/lv_ufs.c

You can have a type (a structure) which can store everything that you need to know to read a file. For example the last read address (the next `read` should continue from there)

 It's up to your implementation how to handle the reading the SPI data but you can store anything you need in the file structure and give it's size as `drv.file_size = sizeof(my_file_t)`

closed 2019-02-01 by kisvegabor