lvgl/lvgl #349: User define keyboard event

heyingjie, 2018-08-04:
HI
    in some cases,we need deal keyboard event by user application,so that we need throw out key code to user application.but in obj signal,key code is not trans to action.
[image]
    so,change 
[image]
to 
[image]
Is that a reasonable modification?

BesitzeRuf, 2018-08-05:
On other hand you can assign your each button with key_code by your own table... also there is free_num and free_ptr variable, which you can set and read for/from any object.

kisvegabor, 2018-08-09:
I added `lv_indev_get_key` in `dev-5.2` branch.

In the action, you can use: 
```c
lv_indev_t * indev = lv_indev_get_act();
uint32_t last_key = lv_indev_get_key(indev);
```
I hope it also helps.