lvgl/lvgl #350: Suggestion about the code style: remove the mixed declarations and code

abiandy, 2018-08-06:
Dear:

When I compile lvgl code with arm-linux-gnueabi-gcc (which has been widely used during embedded linux world), and I got a lot of warnings like:
  .../lvgl/lv_themes/lv_theme_night.c:549: warning: ISO C90 forbids mixed declarations and code
  .../lvgl/lv_themes/lv_theme_night.c:558: warning: ISO C90 forbids mixed declarations and code
  .../lvgl/lv_themes/lv_theme_night.c: In function 'lv_theme_night_init':
  .../lvgl/lv_themes/lv_theme_night.c:594: warning: ISO C90 forbids mixed declarations and code

So shall the code style be completely compatible with  ISO C90 standard? Just as the linux kernel does?
Thanks very much!

kisvegabor, 2018-08-09:
Hi,

C90 has a lot of limitation compared to C99 and every compiler today should support it especially arm-linux-gnueabi-gcc.

You should compile with `-std=c99`

abiandy, 2018-08-09:
Dear:

I will try it soon, and thanks a lot!

kisvegabor, 2018-09-23:
As there was no activity here for a while I close it to keep the issues clear.

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2018-09-23 by kisvegabor