lvgl/lvgl #354: How to display different size fonts within LGVL?

turoksama, 2018-08-09:
Hi there,
   Is there a way to display different size fonts within LGVL?
 Best Regards.

bernpuc, 2018-08-09:
Yes, it is supported. Check first in lv_conf.h to the section FONT USAGE. There is the first place to enable the different sized fonts supplied by lvgl. Next read about Styles in the documentation which specifies how to set up a new font. In addition, there is an online tool to add any TTF font to your project.
See https://littlevgl.com/basics#styles

turoksama, 2018-08-10:
Hi,
    Thanks a lot. It's very helpful!

     I checked the manual and noticed lib-func in lv_font.c : "void lv_font_init(void)"

void lv_font_init(void)
{
....
lv_font_add(&muli_latin_regular_20, NULL);
lv_font_add(&lv_font_symbol_20, &muli_latin_regular_20);
...
}

(the" muli_latin_regular_20 " which is my built-in font)
It registers the font as built-in font(default), am I right or not?

Still, I have some questions about using "style":
What if I need 3(technically N) other different sizes of font(along with symbols) as user font.

"https://littlevgl.com/basics#styles" says:

/*Create a new style for the label*/
static lv_style_t style;
lv_style_copy(&style, &lv_style_plain);
style.text.color = LV_COLOR_BLUE;
**style.text.font = &lv_font_dejavu_40;**   /*Unicode and symbol fonts already assigned by the library*/

lv_obj_t *label;

/*Use ASCII and Unicode letters*/
label = lv_label_create(lv_scr_act(), NULL);
lv_obj_set_pos(label, 20, 20);
lv_label_set_style(label, &style);
lv_label_set_text(label, "aeuois\n"
                         "äéüöíß");

/*Mix text and symbols*/
label = lv_label_create(lv_scr_act(), NULL);
lv_obj_set_pos(label, 20, 100);
**lv_label_set_style(label, &style);**
**lv_label_set_text(label, "Right "SYMBOL_RIGHT);**
In cording to codes above, the height of SYMBOL_RIGHT should be default.
If I need a height of 40, what should I do to link lv_font_symbol_40 along with muli_latin_regular_40(usr font)?
Like this:

lv_style_t style_10, style_30, style_40;

**lv_label_set_style(label, &style_10);**
**lv_label_set_text(label, "Right "SYMBOL_XXX);**//Label text & SYMBOL height 10

**lv_label_set_style(label, &_style_default_);**
**lv_label_set_text(label, "Right "SYMBOL_XXX);**//Label text & SYMBOL height 20

**lv_label_set_style(label, &style_30);**
**lv_label_set_text(label, "Right "SYMBOL_XXX);**//Label text & SYMBOL height 30

**lv_label_set_style(label, &style_40);**
**lv_label_set_text(label, "Right "SYMBOL_XXX);**//Label text & SYMBOL height 40

……

???
Well, I'll also try it out in my board.
Regards.

turoksama, 2018-08-10:
Hi,
you means that call lv_font_add(…)pairs 4 times will link them all,and addressable by the style funtion?
Regards.

kisvegabor, 2018-08-10:
Yes. Here it is how LittelvGL uses this to link a symbol font to an ascii:
https://github.com/littlevgl/lvgl/blob/master/lv_misc/lv_font.c#L107

After is if you use `my_style.text.font = &lv_font_dejavu_20` it will know the symbols too.

turoksama, 2018-08-10:
Hi,
  So, i can make those predefines "enabled" @FONT USAGE section in lv_conf.h
 and the lib will initialize to link them all.
It's very much kind of you, thanks a lot!
By the way, the library is awesome.
Regards.

kisvegabor, 2018-08-10:
Yes, the built-in fonts are linked automatically but you can link your own fonts too as mentioned above: 
`lv_font_add(&lv_font_symbol_30, &muli_latin_regular_30)`

Thank you! :)
Please let us know how it workes for you!

closed 2018-09-18 by kisvegabor