lvgl/lvgl #355: Suggestions for implementing radio buttons

bernpuc, 2018-08-09:
Hi,
I am using a list object to show a list of items and would like to enable radio-button behavior such that only one item at a time can be toggled/selected. Is there a way to do this?

I tried to implement this myself by iterating through the objects in the list and manually setting the state, but I don't know how to distinguish button objects from other child objects.

Any help is appreciated!
Bernard

bernpuc, 2018-08-09:
I've found a simpler solution which is to store the object pointer of the button that is pressed and to untoggle that button manually on the next button press.

kisvegabor, 2018-08-09:
Hi,

I just wanted to suggest the same! :)

closed 2018-08-09 by kisvegabor