lvgl/lvgl #356: Epaper Display driver question

LGKev, 2018-08-11:
Hi,

I have an epaper display that has it's own graphical ram. Currently I can display pixels by:

1. create large array 
2. change data in array
3. push to graphical ram
4. display graphical ram to display

How can I translate the above to work with the buffered drawing of littlevgl? 

I looked at the examples and didn't see a declaration of an image array, and I saw that LV_VDB  is non zero, so I am assuming they are using the virtualized buffer. I am confused by the additional step of the graphical ram loading/ display image. I looked at the ili9431 data sheet and it also has a graphical ram. So the example should be similar. 

Would the correct implementation be something like combining steps 1 through 4 as a single function call, equivalent to ex_disp_flush? so that the end result is a pixel gets put onto the display even if that means updating the entire display?

The epaper I have also supports partial refresh. I could imagine modifying my steps 1 through 4 and instead of updating the entire display, only update the required portion. 

what do you suggest?

kisvegabor, 2018-08-13:
Hi,

In the `disp_flush` function you get an array of pixels and coordinates. The most simple solution is to set the colors in your buffer based on the parameters of `disp_flush`.

However, you can eliminate your buffer iIf you set your display's RAM directly from the buffer in `disp_flush`.

If your display has a special color format, e.g. 1 byte means 8 pixels,  please let me know because there are features to support it as well.

kisvegabor, 2018-09-18:
As there was no activity here for a while  I suppose the issue is solved.

If it's not working feel free the comment here and reopen this issue.

closed 2018-09-18 by kisvegabor