lvgl/lvgl #359: tabview and load action

lidiriel, 2018-08-13:
Hi,

for using the load action `lv_tabview_set_tab_load_action(tabview, tabview_action);` must be placed after the «tab» creation. If the load action is set beetween tabview creation and tab creation lvgl failed.
I think it is not a problem but a little information about this in documentation will be usefull.

bye

kisvegabor, 2018-08-13:
Hi,

I tested it in the `maser` and the `dev-5.2` branch like this and it worked here:
```c
lv_res_t lv_tv_action(lv_obj_t * tv)
{
	printf("loaded\n");
	return LV_RES_OK;
}

...
    lv_obj_t * tv = lv_tabview_create(lv_scr_act(), NULL);

    lv_tabview_set_tab_load_action(tv, lv_tv_action);
    
    lv_tabview_add_tab(tv, "one");
    lv_tabview_add_tab(tv, "two");
    lv_tabview_add_tab(tv, "tree");
```
Can you send a not working example?

kisvegabor, 2018-09-23:
As there was no activity here for a while I close it to keep the issues clear.

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2018-09-23 by kisvegabor