lvgl/lvgl #360: lv_obj_set_parent is missing several NULL checks

nameofuser1, 2018-08-13:
Take a look at lv_obj_set_parent function please:
https://github.com/littlevgl/lvgl/blob/b65b03765d269570b2126391effb8ffe467f4682/lv_core/lv_obj.c#L408

It will produce undefined behavior when passed object does not have a parent and when new parent is set to NULL.

kisvegabor, 2018-08-13:
Hi,

True, it should be checked. 
However, only screens should have `NULL` parent but `lv_obj_set_parent` is not valid for them.

nameofuser1, 2018-08-14:
Oh, I missed that about screens and `set_parent`. Hope documentation will come soon.

Nevertheless it seems to be much safer and user-friendly to check for parameters and call user defined callback in case of wrong one than leave it with undefined behavior. Maybe it worth to add some simple assert function at least for NULL expressions? What do you think?

kisvegabor, 2018-08-15:
I added error check and logging on error in `dev-5.2`: https://github.com/littlevgl/lvgl/blob/dev-5.2/lv_core/lv_obj.c#L437

Thank you for pointing his out!

closed 2018-08-15 by kisvegabor