lvgl/lvgl #361: How to design name/value editing page

bernpuc, 2018-08-13:
Hi All,
I'm trying to design a screen which will contain a number of name/value pairs and the value can be edited. My problem is that I would like the user to be able to click (give focus) to a value field and that action would pop up a keyboard with which the value can be edited.

Is there example code I could look at that does something similar?

I'm thinking the way to do it is to place invisible buttons on top of the text area objects and when the button is clicked, show a keyboard.

Any guidance much appreciated!
-Bernard

kisvegabor, 2018-08-13:
Hi,

I've just added the `lv_ta_set_action(ta, callback)` to `dev-5.2` branch. It works like the button actions.

bernpuc, 2018-08-13:
Thanks. Good to know.

closed 2018-08-13 by bernpuc