lvgl/lvgl #363: unable to show binary images

eugen-vusak, 2018-08-14:
When recreating images tutorial, binary images aren't displayed (path is dispayed as text ). 
I changed 
`    pc_file_t * fp = file_p;        /*Just avoid the confusing casings*/
`
with
`    pc_file_t * fp = (FILE**) file_p;        /*Added confusing casings*/
`
Also I'm on Linux machine not Windows so path used is of different kind.

kisvegabor, 2018-08-15:
Hi,

Did the `  lv_tutorial_image();` work well?

If yes, then maybe you placed the image to a wrong place. If in `fopen` you use a path like `./pic1.bin` then it will be relative to the project's root folder and not the executable `pc_simulator` in the Debug folder.

Can you test it if `fopen` can open the file?

rockcastle, 2018-08-16:
Hi,
I have checked on mine and lv_tutorial_image() is not showing blue_flower.
That is img_bin, right?
What should i did wrong?

eugen-vusak, 2018-08-16:
Hi, 

I basically copy pasted `lv_tutorial_image();` code, because I didn't clone lv_examples git repo. 

Using only `fopen` I'm able to open file without an error so my path is correct

rockcastle, 2018-08-16:
Any sample code please?

eugen-vusak, 2018-08-16:
lv_init();
    fbdev_init();
    /*Add a display the LittlevGL sing the frame buffer driver*/
    lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);
    disp_drv.disp_flush = fbdev_flush;      /*It flushes the internal graphical buffer to the frame buffer*/
    lv_disp_drv_register(&disp_drv);

    lv_obj_t * scr1;
    scr1 = lv_obj_create(NULL, NULL);
    lv_scr_load(scr1);

    /**************************
     * IMAGE FROM BINARY FILE
     **************************/

    /* Add a simple drive to open images from PC*/
    lv_fs_drv_t pcfs_drv;                         /*A driver descriptor*/
    memset(&pcfs_drv, 0, sizeof(lv_fs_drv_t));    /*Initialization*/

    pcfs_drv.file_size = sizeof(pc_file_t);       /*Set up fields...*/
    pcfs_drv.letter = 'P';
    pcfs_drv.open = pcfs_open;
    pcfs_drv.close = pcfs_close;
    pcfs_drv.read = pcfs_read;
    pcfs_drv.seek = pcfs_seek;
    pcfs_drv.tell = pcfs_tell;
    lv_fs_add_drv(&pcfs_drv);

    lv_obj_t * img_bin = lv_img_create(lv_scr_act(), NULL); /*Create an image object*/

    lv_img_set_src(img_bin, "resources/pictures/blue_flower_16.bin");
    lv_obj_set_pos(img_bin, 10, 10);      /*Set the positions*/
    lv_obj_set_drag(img_bin, true);

    std::thread gui_thread([]() {
        while (1) {
            lv_task_handler();
            lv_tick_inc(5);
            usleep(5);
        }
    });
    
    gui_thread.join();
    return 0;

That's my main program. It's written in c++.
And this is the part I added for checking path with fopen:
(sorry for this type of paste, i was unable to use `code` paste without it breaking)

`FILE *fptr;`
`fptr = fopen("resources/pictures/blue_flower_16.bin", "rb"); // this works `
`// fptr = fopen("./resources/pictures/blue_flower_16.bin", "rb"); // this works aswell`
`if (fptr == NULL){`
`    printf("Error!\n");`
`    exit(1);`
`}`
`fclose(fptr);`

kisvegabor, 2018-08-17:
Hi,

Just to start from a common point. Please, everybody, try the `lv_examples` repo first. :)

eugen-vusak, 2018-08-18:
I have now tried lv_examples. 
First problem is that ["lv_examples/lv_tutorial/6_images/lv_tutorial_images.h"](https://github.com/littlevgl/lv_examples/blob/master/lv_tutorial/6_images/lv_tutorial_images.h) is copy pasted lv_tutorial_antialiasing

With that aside, binary image is still not shown. Red flower (source image) and Check symbol are both shown but blue flower (binary image) fails and only prints location of file.

File by it self openable by `fopen` with and without ./ at the beginning.

kisvegabor, 2018-08-18:
The header is really wrong. Thank you!

I tested the `master` branches it works for me.

Please, debug what happens in `pcfs_open` in `lv_tutorial_images.c`. Can it open th file?

kisvegabor, 2018-09-23:
As there was no activity here for a while I close it to keep the issues clear.

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2018-09-23 by kisvegabor