lvgl/lvgl #364: Image scrolling

w521xu, 2018-08-15:
Hi,now I can achieve the image scrolling in a straight line, similar to the list, can you achieve scrolling in a circular path? Thank you

kisvegabor, 2018-08-15:
Hi,

Try this:
```c
    /*Create a Page which is a scrollable object*/
    lv_obj_t * page = lv_page_create(lv_scr_act(), NULL);
    lv_obj_set_size(page, 100, 100);

    /*Create an Image on the page*/
    LV_IMG_DECLARE(grad);
    lv_obj_t * img = lv_img_create(page, NULL);
    lv_img_set_src(img, &grad);             /*Set the source image*/
    lv_img_set_auto_size(img, false);       /*Disable auto auto size*/
    lv_obj_set_size(img, 100, 800);         /*Set size manually. The image will be repeated*/
    lv_page_glue_obj(img, true);            /*"Glue" the image to the page which makes possible to move the page by dragging the image*/
```

I got this:
[image]

w521xu, 2018-08-16:
Hi,

Thank you very much for your answer,but this answer is not what I want. Sorry,I didn't make it clear. The result I want is a cool sliding menu. I put five images in a row, where the image size in the middle is the largest , and the image sizes on both sides are reducde in size to become an arc shape. Then I slide the screen with my finger, and the five images are sequentially scrolled. Finally, I click on the middle image to implement a corresponding function. The GIF image I uploaded is made by TouchGFX, so I want to know if LiitlevGL can do this?

[image]

kisvegabor, 2018-08-17:
Oh, I see.
As image scaling is not supported right now, there isn't a "sliding menu" widget like this. :(

w521xu, 2018-08-20:
Thank you.
 I hope that littlevgl will have the function of sliding menu in the future.

kisvegabor, 2018-08-21:
I hope so :)
As too much prerequisite is missing for this feature right now, I close this issue. Once the image scaling will be added (which is planned) then this idea can be realized.

closed 2018-08-21 by kisvegabor

david-res, 2020-09-24:
@kisvegabor Sorry to be bumping a very old issue, but can this be done in V7.x at all?
I was thinking of using a list or a roller to vertically scroll between a few images, but didn't see an option to do so.

embeddedt, 2020-09-24:
It should be possible with https://docs.lvgl.io/latest/en/html/widgets/img.html#offset.

david-res, 2020-09-24:
And no other way to do it besides animation with the offsets? 
I was hoping the stickiness or the roller “click” would look nice

embeddedt, 2020-09-24:
Thinking about this again, you might be able to use a tileview and create an image object on each tile. I don't know how animated the tileview's scrolling is though.

https://docs.lvgl.io/latest/en/html/widgets/tileview.html#example