lvgl/lvgl #366: Invert label color in monochrome display

IronCaterpillar, 2018-08-15:
Hello! I want to invert the colour of label in monochrome display to highlight it. How can I do it? I've tried playing with styles but didn't succeed.

kisvegabor, 2018-08-15:
Hi,

Please copy here what have you tried so far. It should work with styles.

IronCaterpillar, 2018-08-16:
```
static lv_style_t style1, style2 , style3;
lv_obj_t *label4 = lv_label_create(container, NULL);
lv_style_copy(&style3, &lv_style_plain);
style3.text.color=LV_COLOR_WHITE;
lv_label_set_text(label4, "TestInverted");
style3.body.main_color = LV_COLOR_BLACK;
style3.body.grad_color = LV_COLOR_BLACK;
lv_obj_set_style(label4, &style3);
lv_obj_set_pos(label4, 0, 54);
```
After this the body colour of the label doesn't change to black. The label disappears(because of text.color=white).

kisvegabor, 2018-08-17:
By default, the background is not drawn behind labels. You can enable this with `lv_label_set_body_draw`

Tha background is bigger with `body.hpad` and body.vpad` than the label and `lv_obj_set_pos` sets the position of the label and not the background.

IronCaterpillar, 2018-08-17:
Thanks! It works now.

closed 2018-08-17 by IronCaterpillar

mohamed-kamel, 2025-03-19:
Unfortunately. This no longer works with LVGL 8.3.0, since the ` lv_label_set_body_draw` no longer exists in lv_label.h
@kisvegabor is there an alternative approach ?

mohamed-kamel, 2025-03-19:
Solved, it has to do with the Opacity of the object. Solved like this

```
lv_obj_set_style_bg_color(label, lv_color_hex(0x000000), LV_PART_MAIN);
lv_obj_set_style_bg_opa(label, LV_OPA_COVER, LV_PART_MAIN);
lv_obj_set_style_text_color(screen, lv_color_hex(0xffffff), LV_PART_MAIN);
```