lvgl/lvgl #367: dev_5_2 How to solve this problem?

GddWuliu, 2018-08-16:
#337 
[image]

BesitzeRuf, 2018-08-16:
What are you settings of project in KEIL environment (C, C++)?

kisvegabor, 2018-08-16:
Check this: http://www.keil.com/forum/18454/cast-to-structure-not-working/
As the last but one comments says: "After adding --c99 in "Misc Controls" on "C/C++" Options, it compiles."

GddWuliu, 2018-08-16:
### [image]

GddWuliu, 2018-08-16:
So if I add c99, it still doesn't work

GddWuliu, 2018-08-16:
thanks!
problem was solved .
 After adding --gnu --c99 in "Misc Controls" on "C/C++" Options, it compiles

GddWuliu, 2018-08-16:
thanks!
 problem was solved .
 After adding --gnu  in "Misc Controls" on "C/C++" Options, it compiles