lvgl/lvgl #368: The line obj can not set position

wcyingdream, 2018-08-16:
Hi,
kisvegabor,
I want to set the postion of line obj in an button,but it can not work whether i using "lv_obj_set_pos" or "lv_obj_align".  Can you test it?
`static const lv_point_t point_on_line[] = {{5, 12}, {60, 12}};
    lv_coord_t color_btn_width = 120;
    lv_coord_t color_btn_height = 80;
    /*Test styling*/
    static lv_style_t line_style[9];
    for(i=0;i<9;i++)
    {
        lv_style_copy(&line_style[i], &lv_style_pretty);
        line_style[i].line.color = LV_COLOR_WHITE;
    }
    line_style[0].line.width = 1;
    line_style[1].line.width = 2;
    line_style[2].line.width = 4;
    line_style[3].line.width = 6;
    line_style[4].line.width = 8;
    line_style[5].line.width = 10;
    line_style[6].line.width = 12;
    line_style[7].line.width = 16;
    line_style[8].line.width = 20;

    for(j=0;j<3;j++)
    {
        for(i=0;i<3;i++)
        {
            btn[i+j*3] = lv_btn_create(win, NULL);
            lv_obj_set_size(btn[i+j*3],color_btn_width,color_btn_height);
            lv_obj_set_pos(btn[i+j*3], color_btn_width*i, color_btn_height*j);
            lv_btn_set_style(btn[i+j*3],LV_BTN_STYLE_REL,&btn_style_rel);
            lv_btn_set_style(btn[i+j*3],LV_BTN_STYLE_PR,&btn_style_pr);
            lv_btn_set_action(btn[i+j*3],LV_BTN_ACTION_CLICK,line_btn_click_action);

            line[i+j*3] = lv_line_create(btn[i+j*3],NULL);
//            lv_obj_set_pos(line[i+j*3],25,25);
            lv_line_set_style(line[i+j*3],&line_style[i+j*3]);
            lv_line_set_points(line[i+j*3],point_on_line,2);
            lv_obj_align(line[i+j*3],btn[i+j*3],LV_ALIGN_CENTER,0,0);
        }
    }`

kisvegabor, 2018-08-17:
Hi,

Buttons have align == `LV_ALIGN_CENTER` by default and it postions everything to the center. You need to apply `lv_btn_set_layout(btn, LV_LAYOUT_OFF);` to turn it off.

Sidenote:
Please, use the ` ```c .... ``` ` tag for code block.  I updated it in your comment now)

wcyingdream, 2018-08-18:
Thanks!
I will try it later.

kisvegabor, 2018-09-23:
As there was no activity here for a while  I suppose the issue is solved.

If it's not working feel free the comment here and reopen this issue.

closed 2018-09-23 by kisvegabor