lvgl/lvgl #370: Scrolling in a windows with several objects

sokolbiba, 2018-08-17:
Hi,
I am trying to navigate throughout the objects (buttons,labels,empy objects) in a window. I can see the scrollbars in the screen but I am the screen is not performing any scrolling and so I am not able to see all the objects. Please can you provide some tips how to deal with that ? 
Regards,
Sokol

sokolbiba, 2018-08-17:
Ok, I found that I had to ENABLE animation  in the config file.
👍

kisvegabor, 2018-08-17:
Hi,

Everything should work without animations too. How does your code work now?

sokolbiba, 2018-08-17:
It does not work if I dissable the animation. I can see the focus moving to the next/prev object but the window is not scrolling. While the animation is enabled it works fine.

kisvegabor, 2018-08-17:
It shouldn't scroll automatically by default. Maybe do you use `lv_group_set_focus_cb`?

sokolbiba, 2018-08-17:
Yes, I use it like this :
```
g = lv_group_create
lv_group_set_focus_cb(g, group_focus_cb)
```
in the init function. Then, I use  the navigate function 

```
void navigate(uint8_t direction,lv_group_t * group)
{
	switch (direction)
	{
	case LV_GROUP_KEY_NEXT:
	{

		lv_group_focus_next(group);

		break;
	}
	case LV_GROUP_KEY_PREV:
	{

		lv_group_focus_prev(group);

		break;
	}
	}
}
```
```navigate``` function only moves the focus but not hte scroller.

kisvegabor, 2018-08-18:
I see. It's already fixed in `dev-5.2`. So in the next release, it will work without animations too.

closed 2018-08-18 by kisvegabor