lvgl/lvgl #371: dev 5.2, system fail

sokolbiba, 2018-08-17:
Hi,
I am now trying to implement the ```dev 5.2 branch``` to nxp MCU. What I basically did is:
1- I downloaded the ```dev 5.2 branch``` .
2- Made the configuration in the config file.
3-Excluded the ```lvgl``` library from the project and included the ```dev 5.2 branch```.
Seems that the System us failing at some point. Here is the print log. 
```
Trace: lv_init started 	(../source/lvgl-dev-5.2/lv_core/lv_obj.c #65)
Trace: Screen create started 	(../source/lvgl-dev-5.2/lv_core/lv_obj.c #128)
Info: Screen create ready 	(../source/lvgl-dev-5.2/lv_core/lv_obj.c #180)
Trace: Screen create started 	(../source/lvgl-dev-5.2/lv_core/lv_obj.c #128)
Info: Screen create ready 	(../source/lvgl-dev-5.2/lv_core/lv_obj.c #180)
Trace: Screen create started 	(../source/lvgl-dev-5.2/lv_core/lv_obj.c #128)
Info: Screen create ready 	(../source/lvgl-dev-5.2/lv_core/lv_obj.c #180)
Info: lv_init ready 	(../source/lvgl-dev-5.2/lv_core/lv_obj.c #108)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: display refresh task started 	(../source/lvgl-dev-5.2/lv_core/lv_refr.c #171)
Trace: disp flush  started 	(../source/lvgl-dev-5.2/lv_hal/lv_hal_disp.c #157)
Trace: disp flush ready 	(../source/lvgl-dev-5.2/lv_hal/lv_hal_disp.c #159)
Trace: display refresh task finished 	(../source/lvgl-dev-5.2/lv_core/lv_refr.c #192)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: lv_task_handler ready 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #134)
Trace: lv_task_handler started 	(../source/lvgl-dev-5.2/lv_misc/lv_task.c #59)
Trace: indev task started 	(../source/lvgl-dev-5.2/lv_core/lv_indev.c #268)
Trace: idnev read started 	(../source/lvgl-dev-5.2/lv_hal/lv_hal_indev.c #115)
Trace: idnev read finished 	(../source/lvgl-dev-5.2/lv_hal/lv_hal_indev.c #117)
```
In both cases I am using the FreeRTOS. I will try to implemet the ```dev 5.2 branch``` in a bare metal and I will let you know.
Regards

kisvegabor, 2018-08-18:
Hi,

It seems the library crashes during input device. processing. What happens if you don't register any input device?

If you have only one task which calls `lv_task_handler` periodically then it should be like bare metal configuration. Except for the stack size of the task. 
Do you have other tasks too? E.g. to read the input device to do something with the GUI?

sokolbiba, 2018-08-23:
You are right. Seems that it is crashing during the registration of input deivce. I removed the registration of input device and everything seems to work smoothly. I managed to use the input deivce using interrupts for both push button and the rotary encoder. The following is the interrupt for the push button:
```
void PORTC_IRQHandler(void)
{

	GPIO_PortClearInterruptFlags(GPIOC, 1<<11);
	BaseType_t xHigherPriorityTaskWoken = pdFALSE;

	if(!button_interrupt_flag)
	{
		button_interrupt_flag=1;
	}
	lv_group_send_data(g, LV_GROUP_KEY_ENTER);
	portYIELD_FROM_ISR(xHigherPriorityTaskWoken);
}
```

kisvegabor, 2018-08-23:
So using interrupts it works well with LittlevGL? If yes, what was the previous approach which didn't work?

sokolbiba, 2018-08-24:
I am not sure if the error was because the input device registration. 
Here is how I am doing the registration of input device:
```	lv_indev_drv_t encoder_dev_button;
	lv_indev_drv_init(&encoder_dev_button);
	encoder_dev.type=LV_INDEV_TYPE_BUTTON;
	encoder_dev.read=&read_button;
	encoder_indev=lv_indev_drv_register(&encoder_dev_button);
```
I was not using the rotary encoder, just the push button. Here is the ```read_button``` function:
```
static bool read_button (lv_indev_data_t * data)
{

//	data->btn=lv_obj_get_free_num(btn1);
	if(!GPIO_PinRead(GPIOC, 11)&&button_interrupt_flag)
	{
		button_interrupt_flag=0;
		data->user_data=(void*)0;
		data->state =LV_INDEV_STATE_PR;

	}
	return 0;
}
```

kisvegabor, 2018-08-24:
It seems you are mixing `encoder_dev` and `encoder_dev_button` here:

```c
        lv_indev_drv_init(&encoder_dev_button);
	encoder_dev.type=LV_INDEV_TYPE_BUTTON;         //Should be encoder_dev_button
	encoder_dev.read=&read_button;                 //Should be encoder_dev_button
	encoder_indev=lv_indev_drv_register(&encoder_dev_button);
```