lvgl/lvgl #374: Mini-review of LittlevGL in progress

bkauler, 2018-08-19:
Hi guys, this post is to let you know about my mini review/evaluation, and to ask a question about next step.

Here are my two blog posts:

http://bkhome.org/news/201808/first-go-at-evaluating-littlevgl.html

http://bkhome.org/news/201808/tentative-first-step-framebuffer-with-littlevgl.html

I am doing a couple of other things right now, but in a day or two intend to write part-3 of the review. I was wondering if someone could help me to extend that example Hello World program in part-2, with maybe a button, and with mouse input.

This is to work with the Linux framebuffer, and evdev is available.

If I have done anything wrong with the Makefile, let me know!

kisvegabor, 2018-08-21:
Hi,

Great reviews!

Let's start with the mouse support:

Here is an SDL based mouse driver: https://github.com/littlevgl/lv_drivers/blob/master/indev/mouse.c 
It is initialized in the PC simulator by default. 

Here you find an evdev driver: https://github.com/littlevgl/lv_drivers/blob/master/indev/evdev.c
Keep in mind, AFAIK, in command line mode (if you kill Xorg) the evdev's are not working by default. 

To add a cursor to the GUI modify `hal_init` in `main.c` like this:
```c
    mouse_init();
    lv_indev_drv_t indev_drv;
    lv_indev_drv_init(&indev_drv);          /*Basic initialization*/
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.read = mouse_read;         /*This function will be called periodically (by the library) to get the mouse position and state*/
    lv_indev_t * mouse_indev = lv_indev_drv_register(&indev_drv);

    lv_obj_t * cursor_obj =  lv_img_create(lv_scr_act(), NULL); /*Create an image for the cursor */
    lv_img_set_src(cursor_obj, SYMBOL_POWER);                 /*For simlicity add a built in symbol not an image*/
    lv_indev_set_cursor(mouse_indev, cursor_obj); /* connect the object to the driver*/
```

renegat56, 2018-08-21:
Hi,
I'm just also fighting with mouse support of lvgl in linux framebuffer. So far I managed to write a prog which can display text and values from my sensors in different colors and size on the framebuffer /dev/fb0. Next I like to create buttons and rulers in display and control them with gmp mouse. I know gmp mouse support is working as /dev/input/event3 in my case, because I have written my own program without lvgl, but ncurses, which is an array of a lot of simple buttons, which can be controlled (clicked) with the gmp mouse and initiates certain actions in linux (playing music, display sensor values, showing images, halt, reboot etc.

Now I would like to use lvgl library as gui, because already has a lot of controls and graphics, which I need.
But I'm unable to get the mouse working. I have the cursor, but on click nothing happened. As start point I used following code

```
#include` "lvgl/lvgl.h"
#include "lv_drivers/display/fbdev.h"
#include <unistd.h>
#include <stdio.h>

//Add a display for the LittlevGL using the frame buffer driver
void register_display(void)
{
	lv_disp_drv_t disp_drv;
	lv_disp_drv_init(&disp_drv);
	disp_drv.disp_flush = fbdev_flush;      //It flushes the internal graphical buffer to the frame buffer
	lv_disp_drv_register(&disp_drv);
}

static lv_res_t btn_click_action(lv_obj_t * btn)
{
    uint8_t id = lv_obj_get_free_num(btn);

    printf("Button %d is released\n", id);

    /* The button is released.
     * Make something here */

    return LV_RES_OK; /*Return OK if the button is not deleted*/
}

int main(void)
{
    /*LittlevGL init*/
    lv_init();

    /*Linux frame buffer device init*/
    fbdev_init();

    // get a display
	register_display();
	
	// enable event input
	evdev_init();
	
	// get an input device like mouse
	// register_inputdev();
	
	/*Create a title label*/
	lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);
	lv_label_set_text(label, "Default buttons");
	lv_obj_align(label, NULL, LV_ALIGN_IN_TOP_MID, 0, 5);
	
	/*Create a normal button*/
	lv_obj_t * btn1 = lv_btn_create(lv_scr_act(), NULL);
	lv_cont_set_fit(btn1, true, true); /*Enable resizing horizontally and vertically*/
	lv_obj_align(btn1, label, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
	lv_obj_set_free_num(btn1, 1);   /*Set a unique number for the button*/
	lv_btn_set_action(btn1, LV_BTN_ACTION_CLICK, btn_click_action);
	
	/*Add a label to the button*/
	label = lv_label_create(btn1, NULL);
	lv_label_set_text(label, "Normal");
	
	/*Copy the button and set toggled state. (The release action is copied too)*/
	lv_obj_t * btn2 = lv_btn_create(lv_scr_act(), btn1);
	lv_obj_align(btn2, btn1, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
	lv_btn_set_state(btn2, LV_BTN_STATE_TGL_REL);  /*Set toggled state*/
	lv_obj_set_free_num(btn2, 2);               /*Set a unique number for the button*/
	
	/*Add a label to the toggled button*/
	label = lv_label_create(btn2, NULL);
	lv_label_set_text(label, "Toggled");
	
	/*Copy the button and set inactive state.*/
	lv_obj_t * btn3 = lv_btn_create(lv_scr_act(), btn1);
	lv_obj_align(btn3, btn2, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
	lv_btn_set_state(btn3, LV_BTN_STATE_INA);   /*Set inactive state*/
	lv_obj_set_free_num(btn3, 3);               /*Set a unique number for the button*/
	
	/*Add a label to the inactive button*/
	label = lv_label_create(btn3, NULL);
	lv_label_set_text(label, "Inactive");

	/*Handle LitlevGL tasks (tickless mode)*/
    while(1)
	{
        lv_tick_inc(5);
        lv_task_handler();
        usleep(5000);
    }

    return 0;
}
```
It compiles and display the buttons and gmp mouse cursor, but on click on button, there is no printf from callback function. I'm missing something, but I don't know what. I enabled edev support in lv_drv_conf.h with /dev/input/event3 (because I know it the correct event for my mouse). I added evdev_init(); in main, but still no success. (sorry, if format looks ugly, I'm not sure how to do it correctly)

bkauler, 2018-08-23:
renegat56, kisvegabor,
Regarding getting evdev working without X, do I need to use gmp?

renegat56, in your example, you haven't added mouse-cursor code as shown by kisvegabor, but you get a mouse cursor -- is that because of gmp?

I tried your program, and I get the buttons, but no mouse. But, I don't have gmp.

bkauler, 2018-08-23:
Got it! This works:

```
#include "lvgl/lvgl.h"
#include "lv_drivers/display/fbdev.h"
#include "lv_drivers/indev/evdev.h"
#include <unistd.h>
#include <stdio.h>

//Add a display for the LittlevGL using the frame buffer driver
void register_display(void)
{
	lv_disp_drv_t disp_drv;
	lv_disp_drv_init(&disp_drv);
	disp_drv.disp_flush = fbdev_flush;      //It flushes the internal graphical buffer to the frame buffer
	lv_disp_drv_register(&disp_drv);
}

static lv_res_t btn_click_action(lv_obj_t * btn)
{
    uint8_t id = lv_obj_get_free_num(btn);

    printf("Button %d is released\n", id);

    /* The button is released.
     * Make something here */

    return LV_RES_OK; /*Return OK if the button is not deleted*/
}

int main(void)
{
    /*LittlevGL init*/
    lv_init();

    /*Linux frame buffer device init*/
    fbdev_init();

    // get a display
	register_display();
	
	// enable event input
	evdev_init();
	
	// get an input device like mouse
	lv_indev_drv_t indev_drv;
	lv_indev_drv_init(&indev_drv);
	indev_drv.type = LV_INDEV_TYPE_POINTER;
	indev_drv.read = evdev_read;
//	lv_indev_drv_register(&indev_drv);

    lv_indev_t * mouse_indev = lv_indev_drv_register(&indev_drv);

    lv_obj_t * cursor_obj =  lv_img_create(lv_scr_act(), NULL); /*Create an image for the cursor */
    lv_img_set_src(cursor_obj, SYMBOL_POWER);                 /*For simlicity add a built in symbol not an image*/
    lv_indev_set_cursor(mouse_indev, cursor_obj); /* connect the object to the driver*/

	
	/*Create a title label*/
	lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);
	lv_label_set_text(label, "Default buttons");
	lv_obj_align(label, NULL, LV_ALIGN_IN_TOP_MID, 0, 5);
	
	/*Create a normal button*/
	lv_obj_t * btn1 = lv_btn_create(lv_scr_act(), NULL);
	lv_cont_set_fit(btn1, true, true); /*Enable resizing horizontally and vertically*/
	lv_obj_align(btn1, label, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
	lv_obj_set_free_num(btn1, 1);   /*Set a unique number for the button*/
	lv_btn_set_action(btn1, LV_BTN_ACTION_CLICK, btn_click_action);
	
	/*Add a label to the button*/
	label = lv_label_create(btn1, NULL);
	lv_label_set_text(label, "Normal");
	
	/*Copy the button and set toggled state. (The release action is copied too)*/
	lv_obj_t * btn2 = lv_btn_create(lv_scr_act(), btn1);
	lv_obj_align(btn2, btn1, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
	lv_btn_set_state(btn2, LV_BTN_STATE_TGL_REL);  /*Set toggled state*/
	lv_obj_set_free_num(btn2, 2);               /*Set a unique number for the button*/
	
	/*Add a label to the toggled button*/
	label = lv_label_create(btn2, NULL);
	lv_label_set_text(label, "Toggled");
	
	/*Copy the button and set inactive state.*/
	lv_obj_t * btn3 = lv_btn_create(lv_scr_act(), btn1);
	lv_obj_align(btn3, btn2, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
	lv_btn_set_state(btn3, LV_BTN_STATE_INA);   /*Set inactive state*/
	lv_obj_set_free_num(btn3, 3);               /*Set a unique number for the button*/
	
	/*Add a label to the inactive button*/
	label = lv_label_create(btn3, NULL);
	lv_label_set_text(label, "Inactive");

	/*Handle LitlevGL tasks (tickless mode)*/
    while(1)
	{
        lv_tick_inc(5);
        lv_task_handler();
        usleep(5000);
    }

    return 0;
}

```

bkauler, 2018-08-23:
kisvegabor,
In the above post, I registered the mouse differently from your suggestion. My code is working for me, can you see anything wrong with it?

Also, SYMBOL_POWER is not very appropriate for a mouse pointer! Is there any mouse pointer image available?

Note, I found this page helpful:

http://www.vk3erw.com/index.php/16-software

bkauler, 2018-08-23:
I have written part-3 of my mini-review, created a static executable (linked with uClibc):

http://bkhome.org/news/201808/littlevgl-evaluation-part-3.html

renegat56, 2018-08-23:
@bkauler,
congratulations that you manged to get it working! 
But in my case it does not :-(
I killed gpm mouse process, but still no success. I used your extended code, it compiles, but now I have no mouse cursor anymore (before it was the rectangle one of gpm mouse in virtual terminal mode).

I'm using a raspberry pi 3 with wheezy. Please let me know some details of your setup

- hardware, mouse (USB?)
- linux distribution and version?
- which event did you use for input?
- as I understodd, you do not run a terminal in X server, but directly in vt mode without X server, right?

bkauler, 2018-08-23:
I'm running on a desktop pc, with usb mouse. Intel i5 cpu and Intel video.

EasyOS starts up in text mode, but when the intel kernel driver loads, it changes to graphics framebuffer. So, it is the framebuffer running the screen when I exit from X.

I left the event handler at the default, /dev/input/event0

Running EasyOS 0.9.6 (x86_64), which is a port of Puppy Linux:
http://bkhome.org/news/201808/easyos-version-096-released.html

Yes, EasyOS, like all pups, has a choice in the menu to exit from X to commandline. That's where I test, not in an alternative terminal.

I have Quirky Linux, another of my Puppy-ports, running on my Pi3b, I will test on that, and report back.

bkauler, 2018-08-23:
Yes, it works in the Pi3b.

My Quirky Linux is aarch64, 64-bit, but in all other respects is pretty much like any other Puppy Linux derivative. It is experimental, not publicly released.

No changes, I just copied everything onto the sd-card, booted the Pi, ran make, exited from X and ran "./demo".

renegat56, 2018-08-23:
so finally I made some progress. I stopped gpm mouse server -> rectangle cursor stopped moving with mouse. No other cursor is visible, even I added Gabor's 3 lines for the cursor image. But by blind clicking I found the buttons and could activate the callback function and also the button changed, when I clicked it.

So the only problem which remains is, I can not see a cursor :-( which is quite impractical

@bkauler your change from mouse_init() to edev_init() like in my org code seems resonable, because we use event system of linux

@kisvegabor 
Gabor, please can you give me another advice to see the cursor?

kisvegabor, 2018-08-23:
First of all, sorry for the late answer. Anyway, you made a great progress! :)

@bkauler 
1. I happily hear that it works there! And good to see the next part on your blog!
2. The only difference I see is that you replaced `mouse_init()` with `evdev_init()`. Or have I missed something? Sorry, maybe it was misleading, I thought `mouse_init` should be replaced with your indev init.
3. Real mouse cursor image: You can use the [online image converter](https://littlevgl.com/image-to-c-array) to create a C array from an image. To add the image as cursor:
```c
LV_IMG_DECLARE(my_cursor);
lv_img_set_src(cursor_obj, &my_cursor);
```

@renegat56 
1. Format the code block: You can add syntax highlight like this: ` ```c   ...your code block... ```   `
2. Invisible cursor: please attache your `lv_conf.h`. There are some configs to check.

bkauler, 2018-08-24:
@kisvegabor 
Yes, I changed mouse_init() to evdev_init()

Heh, heh, for now, I am using SYMBOL_GPS, it looks very much like a mouse pointer!

bkauler, 2018-08-24:
@renegat56 
I have bundled my lv_conf.h, lv_drv_conf.h, main.c and Makefile into a tarball, for you to try.
The tarball expands to folder 'wks2'.

[wks2.tar.gz](https://github.com/littlevgl/lvgl/files/2316590/wks2.tar.gz)

renegat56, 2018-08-24:
Hi Gabor,

here is my lv_conf.h. I think I know what is wrong. I haven't enabled the SYMBOL font, which size should I enable? I will check tonight. Am I right or anything else wrong?

```c
/**
 * @file lv_conf.h
 * 
 */

#ifndef LV_CONF_H
#define LV_CONF_H

/*----------------
 * Dynamic memory
 *----------------*/

/* Memory size which will be used by the library
 * to store the graphical objects and other data */
#define LV_MEM_CUSTOM      0                /*1: use custom malloc/free, 0: use the built-in lv_mem_alloc/lv_mem_free*/
#if LV_MEM_CUSTOM == 0
#define LV_MEM_SIZE    (32U * 1024U)        /*Size memory used by `lv_mem_alloc` in bytes (>= 2kB)*/
#define LV_MEM_ATTR                         /*Complier prefix for big array declaration*/
#define LV_MEM_AUTO_DEFRAG  1               /*Automatically defrag on free*/
#else       /*LV_MEM_CUSTOM*/
#define LV_MEM_CUSTOM_INCLUDE <stdlib.h>   /*Header for the dynamic memory function*/
#define LV_MEM_CUSTOM_ALLOC   malloc       /*Wrapper to malloc*/
#define LV_MEM_CUSTOM_FREE    free         /*Wrapper to free*/
#endif     /*LV_MEM_CUSTOM*/

/*===================
   Graphical settings
 *===================*/

/* Horizontal and vertical resolution of the library.*/
#define LV_HOR_RES          (1280/4)
#define LV_VER_RES          (720/2)
#define LV_DPI              200

/* Size of VDB (Virtual Display Buffer: the internal graphics buffer).
 * Required for buffered drawing, opacity and anti-aliasing
 * VDB makes the double buffering, you don't need to deal with it!
 * Typical size: ~1/10 screen */
#define LV_VDB_SIZE         (20 * LV_HOR_RES)  /*Size of VDB in pixel count (1/10 screen size is good for first)*/
#define LV_VDB_ADR          0                  /*Place VDB to a specific address (e.g. in external RAM) (0: allocate automatically into RAM)*/

/* Use two Virtual Display buffers (VDB) parallelize rendering and flushing (optional)
 * The flushing should use DMA to write the frame buffer in the background*/
#define LV_VDB_DOUBLE       0       /*1: Enable the use of 2 VDBs*/
#define LV_VDB2_ADR         0       /*Place VDB2 to a specific address (e.g. in external RAM) (0: allocate automatically into RAM)*/

/* Enable anti-aliasing (lines, and radiuses will be smoothed) */
#define LV_ANTIALIAS        1       /*1: Enable anti-aliasing*/

/*Screen refresh settings*/
#define LV_REFR_PERIOD      50    /*Screen refresh period in milliseconds*/
#define LV_INV_FIFO_SIZE    32    /*The average count of objects on a screen */

/*=================
   Misc. setting
 *=================*/

/*Input device settings*/
#define LV_INDEV_READ_PERIOD            50                     /*Input device read period in milliseconds*/
#define LV_INDEV_POINT_MARKER           0                      /*Mark the pressed points  (required: USE_LV_REAL_DRAW = 1)*/
#define LV_INDEV_DRAG_LIMIT             10                     /*Drag threshold in pixels */
#define LV_INDEV_DRAG_THROW             20                     /*Drag throw slow-down in [%]. Greater value means faster slow-down */
#define LV_INDEV_LONG_PRESS_TIME        400                    /*Long press time in milliseconds*/
#define LV_INDEV_LONG_PRESS_REP_TIME    100                    /*Repeated trigger period in long press [ms] */

[... 174 lines cut ...]

/*Roller (dependencies: lv_ddlist)*/
#define USE_LV_ROLLER    1
#if USE_LV_ROLLER != 0
#define LV_ROLLER_ANIM_TIME     200     /*Focus animation time [ms] (0: no animation)*/
#endif

/*Slider (dependencies: lv_bar)*/
#define USE_LV_SLIDER    1

/*Switch (dependencies: lv_slider)*/
#define USE_LV_SW       1

/*************************
 * Non-user section
 *************************/
#ifdef _MSC_VER                               /* Disable warnings for Visual Studio*/
# define _CRT_SECURE_NO_WARNINGS
#endif

#endif /*LV_CONF_H*/
```

@bkauler 
thanks for the tarball, I will take a look at it tonight

renegat56, 2018-08-24:
Hi,
I couldn't wait and checked. Yes, success!! I enabled the SYMBOL fonts and voila cursor appeared.
I also learned that the gpm mouse conflicts with lvgl.
Sorry to bother you with this stupid mistake and thanks for help!

kisvegabor, 2018-08-24:
@renegat56 
Great! I'm happy to hear it works there! :)

> Sorry to bother you with this stupid mistake and thanks for help!

You didn't bother. Feel free to ask in the future too! :)

closed 2018-08-24 by kisvegabor