lvgl/lvgl #375: Optimisation question

Zaltora, 2018-08-20:
I was looking the tick system when something is questioning me with this system.
Why not replace this code:
```c
static uint32_t sys_time = 0;
static volatile uint8_t tick_irq_flag;
LV_ATTRIBUTE_TICK_INC void lv_tick_inc(uint32_t tick_period)
{
    tick_irq_flag = 0;
    sys_time += tick_period;
}
uint32_t lv_tick_get(void)
{
    uint32_t result;
    do {
        tick_irq_flag = 1;
        result = sys_time;
    } while(!tick_irq_flag);     /*'lv_tick_inc()' clears this flag which can be in an interrupt. Continue until make a non interrupted cycle */

    return result;
}
```
by this code:
```c
static volatile uint32_t sys_time = 0;
LV_ATTRIBUTE_TICK_INC void lv_tick_inc(uint32_t tick_period)
{
    sys_time += tick_period;
}
uint32_t lv_tick_get(void)
{
    return sys_time;
}
```
The initial code is faster i guess ?  `sys_time` usage will not be optimized by the compiler.
why use `do { ... } while` loop,  it prevents what? if an interrupt happen, the `sys_time` is just updated ? 
i am curious.

DeonMarais64, 2018-08-21:
On a 32 bit processor sys_time += tick_period would be an atomic operation, but on an 8 or 16 bit processor it would not be. tick_irq_flag and do{ ... } while loop ensure that the sys_time is unfettered by a tick interrupt when it is sampled.

kisvegabor, 2018-08-21:
Hi,

@DeonMarais64 I just started to write the same answer :)

Zaltora, 2018-08-21:
thank for your answer! To complement:

1)  Why not put `volatile` on `sys_time` since this variable is in an interruption? the fact that it is to modify only in this interuption? I do not have to use the volatile keyword on `sys_time` any more?

2)  Put `volatile` on `sys_time` or not much affects the performance (speed of treatment)?

3)  If the new code is faster, can we integrate it in lvgl with an option (preprocessor) that will be selected according to the architecture (8-16-32-64- ..)?
E.g: `sys_tick` can be used in 64 bits on 64 bits architecture. No need to control whether it overflow either.
We can use a new section in lv_conf to gather all the small optimization options. E.g:
```c
/*=================
   Performance. setting
 *=================*/
#define LV_ATTRIBUTE_TICK_INC                 /* Define a custom attribute to `lv_tick_inc` function */
#define LV_ATTRIBUTE_TASK_HANDLER             /* Define a custom attribute to `lv_task_handler` function */
#define LV_ARCHITECTURE  32
typedef int32_t  lv_pixel_t                               /*   X / Y coordinate  size  */
....
```

kisvegabor, 2018-08-21:
`volatile` is for the optimization. The compiler might optimize out a variable if it's not changed in the code. And if the variable is changed in the interrupt the compiler won't see the change.

As the loop needs to run twice very very rarely your suggestion can have only a very little effect in performance but would add a new configuration option which can be a root of issues if set wrong.

But I maybe I don't see the full picture. In your case, this update would have a bigger effect?

DeonMarais64, 2018-08-21:
Hi,
There is often a tradoff between absolute optimisation and simplicity. IMO they way it is is a good compromise.
In this instance using the volatile keyword for sys_time would do no harm and is unlikely to offer much ITO opsimisation, it would only realy be necesarry if sys_time were a global variable. But that would mean whomever is referencing it would need to provide an atomic modify guard similar to what is already been done for each reference instance in cases where processor word size does not match sys_time's type.

Zaltora, 2018-08-22:
About bit field started at the end of issue #298, the current writing is this:
```c
typedef enum
{
    LV_SB_MODE_OFF 	= 0x0,   	/*Never show scrollbars*/
    LV_SB_MODE_ON  	= 0x1,   	/*Always show scrollbars*/
    LV_SB_MODE_DRAG = 0x2,   	/*Show scrollbars when page is being dragged*/
    LV_SB_MODE_AUTO = 0x3,   	/*Show scrollbars when the scrollable container is large enough to be scrolled*/
	LV_SB_MODE_HIDE = 0x4,	 	/*Hide the scroll bar temporally*/
	LV_SB_MODE_UNHIDE = 0x5,	/*Unhide the previously hidden scrollbar. Recover it's type too*/
} lv_sb_mode_t;

/*Data of page*/
typedef struct
{
    lv_cont_ext_t bg; /*Ext. of ancestor*/
    /*New data for this type */
    lv_obj_t * scrl;            /*The scrollable object on the background*/
    lv_action_t rel_action;     /*Function to call when the page is released*/
    lv_action_t pr_action;      /*Function to call when the page is pressed*/
    struct {
        lv_style_t *style;          /*Style of scrollbars*/
        lv_area_t hor_area;            /*Horizontal scrollbar area relative to the page. (Handled by the library) */
        lv_area_t ver_area;            /*Vertical scrollbar area relative to the page (Handled by the library)*/
        uint8_t hor_draw :1;        /*1: horizontal scrollbar is visible now (Handled by the library)*/
        uint8_t ver_draw :1;        /*1: vertical scrollbar is visible now (Handled by the library)*/
        uint8_t mode     :3;        /*Scrollbar visibility from 'lv_page_sb_mode_t'*/
    } sb;
    uint8_t arrow_scroll :1;		/*1: Enable scrolling with LV_GROUP_KEY_LEFT/RIGHT/UP/DOWN*/
} lv_page_ext_t;
```
the next writing seems more readable/ usefull for you ?  (It is minor) 
```c
enum
{
    LV_SB_MODE_OFF 	= 0x0,   	/*Never show scrollbars*/
    LV_SB_MODE_ON  	= 0x1,   	/*Always show scrollbars*/
    LV_SB_MODE_DRAG = 0x2,   	/*Show scrollbars when page is being dragged*/
    LV_SB_MODE_AUTO = 0x3,   	/*Show scrollbars when the scrollable container is large enough to be scrolled*/
	LV_SB_MODE_HIDE = 0x4,	 	/*Hide the scroll bar temporally*/
	LV_SB_MODE_UNHIDE = 0x5,	/*Unhide the previously hidden scrollbar. Recover it's type too*/
};
typedef uint8_t lv_sb_mode_t;

/*Data of page*/
typedef struct
{
    lv_cont_ext_t bg; /*Ext. of ancestor*/
    /*New data for this type */
    lv_obj_t * scrl;            /*The scrollable object on the background*/
    lv_action_t rel_action;     /*Function to call when the page is released*/
    lv_action_t pr_action;      /*Function to call when the page is pressed*/
    struct {
        lv_style_t *style;          /*Style of scrollbars*/
        lv_area_t hor_area;            /*Horizontal scrollbar area relative to the page. (Handled by the library) */
        lv_area_t ver_area;            /*Vertical scrollbar area relative to the page (Handled by the library)*/
        uint8_t hor_draw :1;        /*1: horizontal scrollbar is visible now (Handled by the library)*/
        uint8_t ver_draw :1;        /*1: vertical scrollbar is visible now (Handled by the library)*/
        lv_sb_mode_t mode     :3;        /*Scrollbar visibility from 'lv_page_sb_mode_t'*/
    } sb;
    uint8_t arrow_scroll :1;		/*1: Enable scrolling with LV_GROUP_KEY_LEFT/RIGHT/UP/DOWN*/
} lv_page_ext_t;
```

C47D, 2018-08-23:
The difference is in the new way to declare the ```lv_sb_mode_t``` enum to be the same size as ```uint8_t```?

Also if i recall correctly bitfields should be at least 2bits wide, with the most significant bit being the sign bit, but don't quote me on that, i'm just thinking loud.

kisvegabor, 2018-08-23:
@Zaltora 
It's a great mixture of the two way of thinking! Congrats! :)
I was thinking about what happens if this way you place two bitfield "below" each other. I thought maybe they will be interpreted as sperate bytes but I was wrong. It works well even if with a mixed `typedef` combo like this:

˙``c
typedef uint8_t enum_8_typedef;
typedef uint16_t enum_16_typedef;

typedef struct {
    uint8_t         field_A :1;
    uint8_t         field_B :2;
    enum_8_typedef  field_C :3;
    enum_16_typedef field_D :3;
    enum_8_typedef  field_E :3;
    enum_16_typedef field_F :3;
}bit_field_struct_t;
```c

It's still 2 bytes (`sizeof(bit_field_struct_t) is `2``). So I think we can adopt this declaration format because it improves the readability while keeps memory usage low!

@C47D
I often use 1 bit wide bitfields and they work for me as simple 0/1 "numbers". I never heard about the sign bit in bit fields. Can show a description of it?

kisvegabor, 2018-08-25:
@C47D 
Thank you!

It says: "a 1-bit signed field on a two's complement machine can only take the values 0 or -1"

I didn't know it but it's quite interesting. However, it's about: **signed field**. So I think until we use unsigned data it"s fine to use 1-bit bitfield too. Do you agree?

Zaltora, 2018-08-25:
It is fine to use 1-bit bitfield, I use this way too. However i will take a look about portability. Using bitfield with register seem not good anymore for me.

I never test to use following `uint8_t` and `uint16_t`. It's not dangerous to do that ? In your case, it is feel to work and use only 2 bytes but i am not sure of the behavior for all compiler.

kisvegabor, 2018-08-26:
> Using bitfield with register seem not good anymore for me.

Why?

> I never test to use following uint8_t and uint16_t. It's not dangerous to do that ? In your case, it is feel to work and use only 2 bytes but i am not sure of the behavior for all compiler.

I also think its dangerous :) Using only `uin8_t` types looks much safer. I just would like to know an extreme mixing of types still works at least in GCC.

C47D, 2018-08-26:
> I didn't know it but it's quite interesting. However, it's about: signed field. So I think until we use unsigned data it"s fine to use 1-bit bitfield too. Do you agree?

I do agree. Anyway, if it worked so far (1bit bitfields) you shouldn't change it.

Zaltora, 2018-08-26:
@kisvegabor  the order is not guarantate. e.g with a 16 bit register for a driver:
```c
typedef union
{
    struct {
        uint16_t cvrf : 1;   // LSB or MSB ?? Compiler dependancy  (Need be LSB to work)
        uint16_t tcf  : 1; 
        uint16_t pvf  : 1; 
        uint16_t wf   : 3;
        uint16_t sf   : 1; 
        uint16_t cf   : 3; 
        uint16_t cen  : 1; 
        uint16_t wen  : 1; 
        uint16_t scc3 : 1;
        uint16_t scc2 : 1; 
        uint16_t scc1 : 1;
        uint16_t      : 1;     //MSB  or LSB ?? compiler dependancy (Need be MSB to work)
    };
    uint16_t mask_register;
} ina3221_mask_t;
```
If you describe a register like this, it will be compiler dependant. In one case, the register description is false.

kisvegabor, 2018-08-28:
In general, it's not a problem if the bitfield declaration and its usage remain in the same code because the compiler will reference the fields as it declared them.

However, if you use it in a message then it's possible that the other system will interpret it differently. And in case of LittlevGL the colors might be problematic because the images are generated externally with a hypothetical bit order.

Anyway, I'd like to keep the current way until somebody reports an issue about it. It seems usually the same order is used in the compilers.

closed 2018-09-18 by kisvegabor

kisvegabor, 2018-09-18:
Thank you for the ideas guys!
I think the final conclusion is the 
```c
 lv_sb_mode_t mode     :3; 
```
bitfield would be better than the current approach. So if you have time and intention, a Pull Request would be welcome. :)