lvgl/lvgl #376: Semicircular scrolling menu

w521xu, 2018-08-23:
Hi,

Thank you for answering my 364th question. Now I have a new idea. Can you create a circular container and create a circular page?"Glue" the image to the page which makes possible to move the page by dragging the image, Implement several images to scroll in a circular page.

w521xu, 2018-08-23:
Similar to this effect.
[image]

kisvegabor, 2018-08-23:
Hi,

It's a very nice simplification of #364 It's really something which can be implemented.

Right now, I'm trying to make v5.2 ready, so I can't implement it myself. However, if you would like to try it, I kindly help you in it! Extra motivation: you would be the first in LittlevGL's community who adds a whole new object type. It would be a big milestone! :)

Initial thoughts: 
- Maybe you already saw that every object type has a `signal functions`. Basically, it's function which defines what to on certain events. E.g. when a button is released it calls its click action and changes its style.
- This object can be derived from `lv_obj` but signal function should be extended. `LV_SIGNAL_PRESSING` should be handled to re-position the children.

What do you think? Are you interested?

w521xu, 2018-08-30:
Hi,
Thank you for agreeing with me, but I am sorry, I am learning LittlevGL for the first time, so I have a lot of knowledge that I don’t know, and I have to be busy with the subject of graduation, I have to leave for a while, but during this time I Also pay attention to LittlevGL.By the way, the library is awesome.
Regards.

kisvegabor, 2018-08-31:
Hi,

Okay, no problem. The graduation is the most important thing now! :)

I think it could be a very good feature but still need to find some time to implement it. So as a reminder I leave this issue opened.

kisvegabor, 2018-10-07:
As there was no activity here for a while I close this issue.

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2018-10-07 by kisvegabor

HareshPrajapati, 2025-03-26:
Any progress on this?
I also want to develop something like this.
Any help will be really appreciated.