lvgl/lvgl #377: Need evdev mouse and keyboard

bkauler, 2018-08-23:
I posted my working program, framebuffer and mouse input:

https://github.com/littlevgl/lvgl/issues/374

evdev is working for the mouse, but not the keyboard. I put a text-entry object into the program, and the cursor blinks, but keyboard is not captured. I have a usb keyboard and usb mouse.

I replaced LV_INDEV_TYPE_POINTER with LV_INDEV_TYPE_KEYPAD, but that only stopped the mouse from working.

Here is the program that I am experimenting with:

```
#include "lvgl/lvgl.h"
#include "lv_drivers/display/fbdev.h"
#include "lv_drivers/indev/evdev.h"
#include <unistd.h>
#include <stdio.h>
#include <stdlib.h>

//Add a display for the LittlevGL using the frame buffer driver
void register_display(void)
{
	lv_disp_drv_t disp_drv;
	lv_disp_drv_init(&disp_drv);
	disp_drv.disp_flush = fbdev_flush;      //It flushes the internal graphical buffer to the frame buffer
	lv_disp_drv_register(&disp_drv);
}

static lv_res_t btn_click_action(lv_obj_t * btn)
{
    uint8_t id = lv_obj_get_free_num(btn);

    printf("Button %d is released\n", id);

    /* The button is released.
     * Make something here */

    return LV_RES_OK; /*Return OK if the button is not deleted*/
}

static lv_res_t btn3_click_action(lv_obj_t * btn)
{
	exit(0);
}

int main(void)
{
    /*LittlevGL init*/
    lv_init();

    /*Linux frame buffer device init*/
    fbdev_init();

    // get a display
	register_display();
	
	// enable event input
	evdev_init();
	
	// get an input device like mouse
	lv_indev_drv_t indev_drv;
	lv_indev_drv_init(&indev_drv);
//	indev_drv.type = LV_INDEV_TYPE_POINTER;
	indev_drv.type = LV_INDEV_TYPE_KEYPAD;
	indev_drv.read = evdev_read;
//	lv_indev_drv_register(&indev_drv);

    lv_indev_t * mouse_indev = lv_indev_drv_register(&indev_drv);

    lv_obj_t * cursor_obj =  lv_img_create(lv_scr_act(), NULL); /*Create an image for the cursor */
    lv_img_set_src(cursor_obj, SYMBOL_GPS);                 /*For simlicity add a built in symbol not an image*/
    lv_indev_set_cursor(mouse_indev, cursor_obj); /* connect the object to the driver*/

	
	/*Create a title label*/
	lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);
	lv_label_set_text(label, "Default buttons");
	lv_obj_align(label, NULL, LV_ALIGN_IN_TOP_MID, 0, 5);
	
	/*Create a normal button*/
	lv_obj_t * btn1 = lv_btn_create(lv_scr_act(), NULL);
	lv_cont_set_fit(btn1, true, true); /*Enable resizing horizontally and vertically*/
	lv_obj_align(btn1, label, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
	lv_obj_set_free_num(btn1, 1);   /*Set a unique number for the button*/
	lv_btn_set_action(btn1, LV_BTN_ACTION_CLICK, btn_click_action);
	
	/*Add a label to the button*/
	label = lv_label_create(btn1, NULL);
	lv_label_set_text(label, "Normal");
	
	/*Copy the button and set toggled state. (The release action is copied too)*/
	lv_obj_t * btn2 = lv_btn_create(lv_scr_act(), btn1);
	lv_obj_align(btn2, btn1, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
	lv_btn_set_state(btn2, LV_BTN_STATE_TGL_REL);  /*Set toggled state*/
	lv_obj_set_free_num(btn2, 2);               /*Set a unique number for the button*/
	
	/*Add a label to the toggled button*/
	label = lv_label_create(btn2, NULL);
	lv_label_set_text(label, "Toggled");
	
	/*Create a one lined text area with password mode*/
	lv_obj_t * ta2 = lv_ta_create(lv_scr_act(), NULL);
	lv_obj_set_size(ta2, 100, 20);
	lv_obj_align(ta2, btn2, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
	//lv_ta_set_style(ta2,LV_TA_STYLE_BG, &style_bg);                     /*Apply the background style*/
	lv_ta_set_one_line(ta2, true);
	lv_ta_set_cursor_type(ta2, LV_CURSOR_LINE);
	lv_ta_set_pwd_mode(ta2, true);
	lv_ta_set_text(ta2, "Password");
	
	/*OK button to exit*/
	lv_obj_t * btn3 = lv_btn_create(lv_scr_act(), btn1);
	lv_obj_align(btn3, ta2, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
	lv_btn_set_action(btn3, LV_BTN_ACTION_CLICK, btn3_click_action);
	label = lv_label_create(btn3, NULL);
	lv_label_set_text(label, "OK");

	/*Handle LitlevGL tasks (tickless mode)*/
    while(1)
	{
        lv_tick_inc(5);
        lv_task_handler();
        usleep(5000);
    }

    return 0;
}

```

kisvegabor, 2018-08-24:
Hi,

Working with a keyboard is not so simple because in this case there has to be an object which focused and only this object should receive the pressed keys. 
To handle this objcets should be added to an `lv_group`. 

Please read this: to lear more about the groups 
https://littlevgl.com/basics#touchad-less-navigation

And check these examples: 
https://github.com/littlevgl/lv_examples/blob/master/lv_tests/lv_test_group/lv_test_group.c
https://github.com/littlevgl/lv_examples/blob/master/lv_tutorial/10_keyboard/lv_tutorial_keyboard.c

bkauler, 2018-08-25:
I have studied the docs and examples, and as far as I can see, usb keyboard input via evdev does not work. I am using 5.1.1.

In the code below, there is only mouse input. The keyboard is not captured.

```
#include "lvgl/lvgl.h"
#include "lv_drivers/display/fbdev.h"
#include "lv_drivers/indev/evdev.h"
#include <unistd.h>
#include <stdio.h>
#include <stdlib.h>

//Add a display for the LittlevGL using the frame buffer driver
void register_display(void)
{
	lv_disp_drv_t disp_drv;
	lv_disp_drv_init(&disp_drv);
	disp_drv.disp_flush = fbdev_flush;      //It flushes the internal graphical buffer to the frame buffer
	lv_disp_drv_register(&disp_drv);
}

int main(void)
{
    /*LittlevGL init*/
    lv_init();

    /*Linux frame buffer device init*/
    fbdev_init();

    // get a display
	register_display();
	
	// enable event input
	evdev_init();
	
	/*Create a title label*/
	lv_obj_t *label1 = lv_label_create(lv_scr_act(), NULL);
	lv_label_set_text(label1, "Title text her");
	lv_obj_align(label1, NULL, LV_ALIGN_IN_TOP_MID, 0, 5);
	
	/*Create a one lined text area with password mode*/
	lv_obj_t * ta2 = lv_ta_create(lv_scr_act(), NULL);
	lv_obj_set_size(ta2, 100, 20);
	lv_obj_align(ta2, label1, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
	lv_ta_set_one_line(ta2, true);
	lv_ta_set_cursor_type(ta2, LV_CURSOR_LINE);
	lv_ta_set_pwd_mode(ta2, true);
	lv_ta_set_text(ta2, "");
	
	/*create a group for keyboard input*/
	lv_group_t *group2 = lv_group_create();
	lv_group_add_obj(group2, ta2);
	
	/*evdev to handle kb input*/
	lv_indev_drv_t kb_drv;
	lv_indev_drv_init(&kb_drv);
	kb_drv.type = LV_INDEV_TYPE_KEYPAD;
	kb_drv.read = evdev_read;
	
	/*assign kb to group*/
	lv_indev_t *kb_indev = lv_indev_drv_register(&kb_drv);
	lv_indev_set_group(kb_indev, group2);
	/*set focus on ta2*/
	lv_group_focus_obj(ta2);
	lv_group_focus_freeze(group2, true);
	
	/*Handle LitlevGL tasks (tickless mode)*/
    while(1)
	{
        lv_tick_inc(5);
        lv_task_handler();
        usleep(5000);
    }

    return 0;
}

bkauler, 2018-08-26:
This is neat, utility named 'skeylogger' scans /proc/bus/input/devices and automatically finds the keyboard event#, then logs all key presses:

https://github.com/gsingh93/simple-key-logger

If LVGL has no way to input from a keyboard via evdev, perhaps this little utility could be incorporated?

Note, that automatic detection could probably be also applied to a mouse, so you wouldn't have to hard-code /dev/input/event#

kisvegabor, 2018-08-26:
Hi,

Yes, `evdev` supports only mouse or touchpad right now.
However, it would be great to have an `evdev_read_pointer` for mouse or touchpad and an `evdev_read_keyboard` for keyboard.

Do you plan to try the utility you mentioned with LittelvGL?

bkauler, 2018-08-27:
Well, I started to think about it, but I am a beginner with LVGL and don't really know how to integrate the simple-key-logger code properly.

I did think that perhaps `lv_drv-conf.h` could be modified, supporting backwards compatibility:

```
/*-------------------------------------------------
 * Keyboard, Mouse or touchpad as evdev interface (for Linux based systems)
 *------------------------------------------------*/
#define USE_EVDEV    1
#if USE_EVDEV
 /*hardcoded event#, for pointer i/p only...*/
 #define EVDEV_NAME   "" /*ex: /dev/input/event0*/
 /*if any of these defined, event# is autodetected...*/
 #define USE_EVDEV_KEYBOARD  1
 #define USE_EVDEV_MOUSE     1 /*mouse and touchpad*/
#endif

```
Then in `evdev.c`, something like this, for function evdev_init():

```
/**********************
 *  STATIC VARIABLES
 **********************/
int evdev_fd = -1;
int evdev_root_x;
int evdev_root_y;
int evdev_button;

int evdev_kbd_fd = -1;
int evdev_mouse0_fd = -1;
int evdev_mouse1_fd = -1;
char event_kbd[9] = "";
char event_mouse0[9] = "";
char event_mouse1[9] = "";

/**
 * Initialize the evdev interface
 * note, instead of running a shell script, in future could use this C code:
 * https://github.com/kernc/logkeys/issues/89
 */
void evdev_init(void)
{
    
    #if USE_EVDEV_KEYBOARD == 1
       char evdev_name_kbd[20] = "/dev/input/";
       /*note, instead of EV=120013, could test for "Handlers=kbd "*/
       char *command1 = "cat /proc/bus/input/devices | tr -d '|' | sed -e 's%^$%|%' | tr '\n' ' ' | tr '|' '\n' | grep 'input0 .*EV=120013 ' | head -n 1 | grep -Eo 'event[0-9]+'";
       FILE *pipe1 = popen(command1, "r");
       if (pipe1 != NULL) {
           fgets(event_kbd, 9, pipe1); /*ex: event1*/
           pclose(pipe1);
           if (event_kbd[0] != 0 ) {
               strcat(evdev_name_kbd, event_kbd);
               evdev_kbd_fd = open(evdev_name_kbd, O_RDWR|O_NOCTTY|O_NDELAY);
               if (evdev_kbd_fd != -1) {
                   fcntl(evdev_kbd_fd, F_SETFL, O_ASYNC|O_NONBLOCK);
               }
	       }
       }
    #endif
    
    #if USE_EVDEV_MOUSE == 1
       /*detect mouse0*/
       char evdev_name_mouse0[20] = "/dev/input/";
       char *command2 = "cat /proc/bus/input/devices | tr -d '|' | sed -e 's%^$%|%' | tr '\n' ' ' | tr '|' '\n' | grep 'input0 .*Handlers=mouse0 ' | head -n 1 | grep -Eo 'event[0-9]+'";
       FILE *pipe2 = popen(command2, "r");
       if (pipe2 != NULL) {
           fgets(event_mouse0, 9, pipe2); /*ex: event0*/
           pclose(pipe2);
           if (event_mouse0[0] != 0) {
               strcat(evdev_name_mouse0, event_mouse0);
               evdev_mouse0_fd = open(evdev_name_mouse0, O_RDWR|O_NOCTTY|O_NDELAY);
               if (evdev_mouse0_fd != -1) {
                   fcntl(evdev_mouse0_fd, F_SETFL, O_ASYNC|O_NONBLOCK);
               }
	       }
       }
       /*detect mouse1*/
       char evdev_name_mouse1[20] = "/dev/input/";
       char *command3 = "cat /proc/bus/input/devices | tr -d '|' | sed -e 's%^$%|%' | tr '\n' ' ' | tr '|' '\n' | grep 'input0 .*Handlers=mouse1 ' | head -n 1 | grep -Eo 'event[0-9]+'";
       FILE *pipe3 = popen(command3, "r");
       if (pipe3 != NULL) {
           fgets(event_mouse1, 9, pipe3); /*ex: event2*/
           pclose(pipe3);
           if (event_mouse1[0] != 0) {
               strcat(evdev_name_mouse1, event_mouse1);
               evdev_mouse1_fd = open(evdev_name_mouse1, O_RDWR|O_NOCTTY|O_NDELAY);
               if (evdev_mouse1_fd != -1) {
                   fcntl(evdev_mouse1_fd, F_SETFL, O_ASYNC|O_NONBLOCK);
               }
	       }
       }
    #endif
    
    /*old method with hard-coded event#*/
    if  (USE_EVDEV_KEYBOARD!=1 & USE_EVDEV_MOUSE!=1) {
        evdev_fd = open(EVDEV_NAME, O_RDWR|O_NOCTTY|O_NDELAY);
        if (evdev_fd != -1) {
            fcntl(evdev_fd, F_SETFL, O_ASYNC|O_NONBLOCK);
        }
    }

    /*error if no input devices found*/
    if (evdev_fd==-1 & evdev_kbd_fd==-1 & evdev_mouse0_fd==-1 & evdev_mouse1_fd==-1) }
        perror("unable open evdev interface");
        return;
    }
    
    evdev_root_x = 0;
    evdev_root_y = 0;
    evdev_button = LV_INDEV_STATE_REL;
}

```
...or something like that!

To run on a system that might have both touchpad and usb mouse, they should work simultaneously. A `evdev_read_mouse()` would have to read both of the "mouse0" and "mouse1" event interfaces, if they exist.

Then there would have to be a `evdev_read_kbd()`, could import the code from simple-key-logger.

It's a bit too much for me at this stage.

kisvegabor, 2018-08-28:
Thank you for the code!

It's a little bit too much for me as well because, although I'm using Linux, I don't have much experience with these things yet.

Anyway, I leave this open for a while. Hopefully, somebody will help to make it ready!

kisvegabor, 2018-10-07:
As there was no activity here for a while I close this issue.

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2018-10-07 by kisvegabor

vipin65, 2019-02-11:
Hi
I am facing similar issue with event dev under linux. In my case I am getting key but some of key missed. Have to press key many times. 
I am wondering why key skipped. i studied SDL keyboard, it uses separate function keyboard_handler(&event). I am confuse need to execute keyboard_handler(&event) in separate pthread under linux ?

kisvegabor, 2019-02-12:
Have look at the current `evdev` driver: https://github.com/littlevgl/lv_drivers/blob/master/indev/evdev.c
It's for a mouse / touchpad but you can get the idea. 

In a `while()` cycle it ready all data from the driver. Do you do something similar?

To debug your issue, please `printf` the events and keys to see where is the problem.

vipin65, 2019-02-12:
Thanks for prompt reply. 
I get key in evdev_read while loop ( printf show key code ) but some time GUI (application) do not respond.
It looks some other function eat up key. Can i get some document how keys processed in lvgl.
I tried to find document on site but could not get enough information.
Regards
Vipin

kisvegabor, 2019-02-14:
As the PC simulator works well with the SDL driver I think the issue is in your evdev driver. Can you share it?

vipin65, 2019-02-14:
Hi
We resolve problem. It was with lv_tick_inc() and lv_task_handler() handling.
We were calling both in same while loop. 
Not working:
while(1)
	{
		lv_tick_inc(1);
		lv_task_handler();
		usleep(1000);
	}
Working:
int count=0;
while(1)
	{
               if(count++>50)
                     { 
                     count=0;    
		lv_tick_inc(1);
                      }
		lv_task_handler();
		usleep(1000);
	}
Can we call lv_tick_inc(1) in separate thread to get more accurate timing ?
Regards
Vipin

embeddedt, 2019-02-14:
@vipin65 You **need** to call `lv_tick_inc` in a separate thread. Otherwise you'll run into problems.

kisvegabor, 2019-02-15:
It strange because now you call `lv_tick_inc(1);` in every 50 ms. So the library thinks that time passes 50 times slower.

Swapping `lv_tick_inc` and `lv_task_handler` seems more logical to me. 

Anyway, once you have an OS calling `lv_tick_inc` from a thread is better. You can use 5 ms sleep too to let the OS breath:
```c     
while(1) {
  lv_tick_inc(5);  
  usleep(5000);
}
```