lvgl/lvgl #378: Horizontal lv_list

phoenix2000, 2018-08-24:
Hi
I couldn't find the attribute for displaying lv_list horizontally.
How is it possible?

kisvegabor, 2018-08-24:
Hi,

It is not supported directly right now but you can do it with a little hack:

```c
	lv_obj_t * list = lv_list_create(lv_scr_act(), NULL);
	lv_obj_set_width(list, 115);			/*It will affect buttons's width*/

	lv_list_add(list, NULL, "first", NULL);
	lv_list_add(list, NULL, "second", NULL);
	lv_list_add(list, NULL, "third", NULL);
	lv_list_add(list, NULL, "fourth", NULL);

       /*Here is teh hack...*/
	lv_page_set_scrl_layout(list, LV_LAYOUT_ROW_M);
	lv_page_set_scrl_fit(list, true, true);
	lv_obj_set_width(list, 320);				/*It will affect the list's width*/
```

And in `lv_list.c` you need to add a `return` to the very beginning of `refr_btn_width()` function.

It resulted me:
[image]
[image]

phoenix2000, 2018-08-25:
Thanks, dear kisvegabor for your reply.
But unfortunately, it didn't work like this. don't you miss anything?
It shows only one button(first one) in the list.

kisvegabor, 2018-08-26:
Hi,

Please copy your current code here. `lv_page_set_scrl_fit(list, false, true);` should not work 

It worked for me to add a scrollbar. It should be after all buttons are added:
```c
	lv_cont_set_fit(list, false, true);
	lv_list_set_sb_mode(list, LV_SB_MODE_AUTO);
```

Sidenote:
If add code block to comment, please use put is between ` ```c .... ``` ` tags.
More info here: https://guides.github.com/features/mastering-markdown/

kisvegabor, 2018-10-07:
As there was no activity here for a while I suppose the issue is solved.

If it's not working feel free the comment here and reopen this issue.

closed 2018-10-07 by kisvegabor

phoenix2000, 2018-10-07:
Hi 
I prefer to use page object in order to make this. My hands are open more in comparison to using list object. for example, I used image button for icons I needed.
PS: Anyway, thanks, dear Gabor.  As before I see you so kind in solving the issues.

kisvegabor, 2018-10-07:
Your welcome!
I'M glad to hear that you found a solution which fits your needs!