lvgl/lvgl #380: How make it work?

prsolucoes, 2018-08-25:
How make it work?

I have download lvgl and download lv_examples inside lvgl. But how i can run it into my raspberry pi 3?

In connecting to my raspberry pi 3 using SSH. I want run the examples, but there is nothing like a Makefile to build and run.

There is not documentation. The only decent tutorial that i found is a video that use eclipse and pc simulator.

Thanks.

kisvegabor, 2018-08-26:
Basically, you can use the makefile from the PC simulator project.

Please read these: 
http://www.vk3erw.com/index.php/16-software
http://bkhome.org/news/201808/first-go-at-evaluating-littlevgl.html

prsolucoes, 2018-08-26:
Hi,

Thanks for fast answer.

I make it work using pc_simulator into my raspberry pi 3. Maybe you can add it to readme after install sdl2:

```
git clone https://github.com/littlevgl/pc_simulator.git
cd pc_simulator
make
./demo
```

But i have another question. Mouse and my touch LCD is not working.
This is my LCD:

https://www.aliexpress.com/item/New-Raspberry-Pi-3-Display-HDMI-7-Inch-1024X600-LCD-with-Touch-Screen-Monitor-for-Raspberry/32682184439.html?spm=a2g0s.9042311.0.0.615c4c4d4Hemur

Can you help me config it?

kisvegabor, 2018-08-26:
Hi,

I updated the readme and porting guide to tell `LV_HOR_RES`, `LV_VER_RES` and `LV_COLOR_DEPTH` should be checked and adjusted. Thank you for pointing out this out.

You should use the `evdev` driver instead of the `mouse` on Raspberry.

prsolucoes, 2018-08-26:
I have configured it. Also you can add it to readme. 

**How to enable touch**

1 - Open "lv_drv_conf.h". Add lines or edit if already exists to be:

```
#define USE_EVDEV 1
#if USE_EVDEV
#define EVDEV_NAME   "/dev/input/event0"
#endif
```

2 - Open "main.c" and add include:

```
#include "lv_drivers/indev/evdev.h"
```

3 - Open "main.c" and inside function "hal_init" add:

```
evdev_init();
lv_indev_drv_t indev_drv_touch;
lv_indev_drv_init(&indev_drv_touch);
indev_drv_touch.type = LV_INDEV_TYPE_POINTER;
indev_drv_touch.read = evdev_read;lv_indev_drv_register(&indev_drv_touch);
```

4 - Make the project and execute

```
make
./demo
```

prsolucoes, 2018-08-27:
Hi,

I change from:

```
#define EVDEV_NAME   "/dev/input/event0"
```

To:

```
#define EVDEV_NAME   "/dev/input/event2"
```

Because when i type "evtest" i got:

```
Available devices:
/dev/input/event0:	USB USB Keykoard
/dev/input/event1:	USB USB Keykoard
/dev/input/event2:	深圳市全动电子技术有限公司 ByQDtech 触控USB鼠标
```

So when i use on event2 it works.

My only problem now is:

1 - The delay between the touch recognise, because i need press between 300ms to it really press the button or other component, its not realtime.

2 - The mouse arrow still appear, but i dont have/use mouse.

kisvegabor, 2018-08-28:
Good to hear you made it work.
I will add info about the use of `evdev` and the `evtest` tools.

1. You should test with `printf` what happens. The info comes slowly from the driver or LittlevGL processes it slowly?

2. As you program is only an application and if Raspberry draws a mouse cursor by default you should find the  (outside of LittelvGL) to hide it

kisvegabor, 2018-09-25:
As there was no activity here for a while  I suppose the issue is solved.

If it's not working feel free the comment here and reopen this issue.

closed 2018-09-25 by kisvegabor