lvgl/lvgl #383: Message box usage question

TMarkivULCRobotics, 2018-08-27:
Trying to use message boxes and found that by default when message box is shown - all other content will still work. So I have two questions:
1 How to disable everything on the screen except message box?  I mean disable clicking for all objects except message box.
2 How to close message box from action function? The message box is shown even after button press, and it doesn't matter what value was returned from the message box action function (LV_RES_INV or LV_RES_OK).

Could you help me with these questions?

kisvegabor, 2018-08-28:
Hi,

1. I usually do it like this:

```c
static lv_style_t gray_style;
lv_style_copy(&gray_style, &lv_style_plain);
gray_style.body.opa = LV_OPA_70;
gray_style.body.main_color = LV_COLOR_HEX3(0x111);
gray_style.body.grad_color = LV_COLOR_HEX3(0x111);

lv_obj_t * gray_bg = lv_obj_create(lv_scr_act(), NULL);
lv_obj_set_style(gray_bg, &gray_style);
lv_obj_set_size(gray_bg, LV_HOR_RES, LV_VER_RES);

static const char * btns[] = {"Yes", "No", ""};

lv_obj_t * mbox = lv_mbox_create(gray_bg, NULL);
lv_mbox_set_text(mbox, "Are you sure you want to power off?");
lv_mbox_add_btns(mbox, btns, poweroff_mbox_action);
lv_obj_align(mbox, NULL, LV_ALIGN_CENTER, 0, 0);

....

static lv_res_t poweroff_mbox_action(lv_obj_t * btns, const char * txt)
{
	lv_obj_t * mbox = lv_mbox_get_from_btn(btns);
	lv_obj_t * gray_bg = lv_obj_get_parent(mbox);
	lv_obj_del(gray_bg);

	return LV_RES_INV;
}
```

2. See `poweroff_mbox_action`

TMarkivULCRobotics, 2018-08-28:
Thank you very much for your answer and the code!

closed 2018-08-28 by TMarkivULCRobotics