lvgl/lvgl #385: teensy 3.2 and images

epikao, 2018-08-28:
Hello,

I have difficult to integrate an image in my arduino / teensy 3.2 project.
I have created an image with the online tool:
```
_For C arrays
Copy the result C file into your LittlevGL project
In a C file of your application declare the image as: extern const lv_img_t my_image_name; or LV_IMG_DECLARE(my_image_name);
Set the image for an lv_img object: lv_img_set_src(img1, &my_image_name);_
```

so now I have a picture.c.
where can I place that file? In the project folder with the ino-file? I'm not sure because then I always get an error.

additional, which code has to be added??, I add follow:
```
LV_IMG_DECLARE(picture);

  lv_obj_t * img_src = lv_img_create(lv_scr_act(), NULL); /*Crate an image object*/
  lv_img_set_src(img_src, &picture);  /*Set the created file as image (a red fl  ower)*/
  lv_obj_set_pos(img_src, 0, 0);      /*Set the positions*/
  lv_obj_set_drag(img_src, true);

```

kisvegabor, 2018-08-28:
Hi,

This part is correct:
```c
  LV_IMG_DECLARE(picture);

  lv_obj_t * img_src = lv_img_create(lv_scr_act(), NULL); /*Crate an image object*/
  lv_img_set_src(img_src, &picture);  /*Set the created file as image (a red fl  ower)*/
  lv_obj_set_pos(img_src, 0, 0);      /*Set the positions*/
  lv_obj_set_drag(img_src, true);
```

However, because Arduino has limited options to set the linked and compiler settings, the best place I found to place your image is `libraries/littelvgl/src`. You need to modify the `include`s in the beginning of the converted files to this:
```c
#include <stdint.h>
#include "../lv_conf.h"
#include "lv_draw/lv_draw.h"
```

I can't find better solution right now. :(
Is it working?

epikao, 2018-08-29:
that works, thank you very much 👍 👍

closed 2018-08-29 by epikao