lvgl/lvgl #386: 1 bit per pixel grayscale tricks

Krastanov, 2018-08-28:
If the DPI is high enough, I have seen implementation which use 1 bpp but still provide some form of grayscale, but alternating pixels.

For instance let's mark white pixels with "." and black pixels with "#". Then 50% gray is:

```
.#.#
#.#.
```

If the DPI is high enough, you can even implement other patterns for other levels of grayscale.

The Pebble smart watch used to do that.

The drawing can easily be done inside of the new  app vdb writing functions: something like `if color is grey and x%2; then draw black; else draw white`. However, what should the color type be for this?

In other words, how do you keep your frame buffer at 1 bit per pixel, but have more than two colors (black, white, and gray, for instance)? The final value of the pixel bit will depend not only on the color, but on the coordinate of the pixel as well.

kisvegabor, 2018-08-31:
Hi,

This technique is called dithering. I wrote a post about it earlier: https://littlevgl.com/blog/9/how-to-use-smooth-images-with-16-bit-color-depth-applying-dithering
At the bottom there is a section about black and white dithering.

kisvegabor, 2018-10-07:
As there was no activity here for a while I close this issue.

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2018-10-07 by kisvegabor