lvgl/lvgl #387: change color on simple objects?

epikao, 2018-08-29:
Hello, 
I just have difficults with I think a very simple issue... I can't find an example how to change the color of an simple object (square, rectangle). 
Follow code gives me an error:
 ```
  static lv_style_t style_obj1;
  style_obj1.obj.color = LV_COLOR_MAKE(0xFF, 0x00, 0x00);

  lv_obj_t * obj1 = lv_obj_create(lv_scr_act(), NULL);/*Create an object */
  lv_obj_set_size(obj1, 50, 50);            /*Set the size of the object*/
  lv_obj_set_style(obj1, &style_obj1);
  lv_obj_set_pos(obj1, 10, 10);               /*Set the position of the  object*/
```

**BTW: Can I also create circles?**

kisvegabor, 2018-08-30:
Hi,

The objects can have a gradient so you should set `body.main_color` and `body.grad_color`.
See this for documentation: https://littlevgl.com/basics#styles

1. Circle: `body.radius = LV_RADIUS_CIRCLE`

2. To convert tumbers to text: 
```c
int num = 100;
char buf[32];
sprintf(buf, "Value: %d", num);
lv_label_set_text(label, buf);
```

epikao, 2018-08-30:
> The objects can have a gradient so you should set body.main_color and body.grad_color.
> See this for documentation: https://littlevgl.com/basics#styles

thanks, I already tried that, but then the object disappears...

kisvegabor, 2018-08-31:
It's because the fields of `style_obj1` variable are not initialized. You should give it an initial value from a built-in style.
E.g. `lv_style_copy(&style_obj1, &lv_style_plain_color)`;

epikao, 2018-09-03:
> E.g. lv_style_copy(&style_obj1, &lv_style_plain_color);

ok. but now my square has a color gradient from red to blue.
I only want have a simple square with red color?, see below my code:
```
  static lv_style_t style_obj1;
  lv_style_copy(&style_obj1, &lv_style_plain_color);
  style_obj1.body.main_color = LV_COLOR_MAKE(0xFF, 0x00, 0x00);

  lv_obj_t * obj1 = lv_obj_create(lv_scr_act(), NULL);/*Create an object */
  lv_obj_set_size(obj1, 50, 50);            /*Set the size of the object*/
  lv_obj_set_style(obj1, &style_obj1);
  lv_obj_set_pos(obj1, 10, 10);               /*Set the position of the  object*/
  //void lv_obj_refresh_style(lv_obj_t * obj);
```

kisvegabor, 2018-09-03:
Yes, `main_color` and `grad_color` should be set as well.

closed 2018-09-03 by kisvegabor