lvgl/lvgl #388: raspberry pi 3 and littlegvl with framebuffer

epikao, 2018-08-30:
Hello,
according my attached image you can see left side the "linux_frame_buffer" folder and structure.
I added in the empty folders "lv_drivers, Lv_examples, lvgl" the corresponding files.
Then I've tried to compile with "gcc -Wall -phtread -o main main.c and I got some errors.

First I needed to change in fbdev.h "#include lvgl/lv_misc/lv_color.h" to "#include ../../lvgl/lv_misc/lv_color.h" + in demo.h  #include "../../../lv_ex_conf.h" to  #include "../../../lv_drv_conf.h" //add PK.

But I still have an error as you see, whats wrong?
[image]

kisvegabor, 2018-08-30:
Hi,

With `gcc -Wall -phtread -o main main.c` only the `main.c` will be compiled. You need to compile the lvgl files and link them together. You should use a Makefile instead. 

I suggest cloning the newest version of the three repositories (linux_frame_buffer is not up to date) and use the Make file from the pc_simulator.

You can also read this: http://www.vk3erw.com/index.php/16-software

epikao, 2018-08-30:
ok I'm trying that now, unfortunately is not explained how to handle that makefile (I never used before a makefile)... so how start the compilation with the makefile?

epikao, 2018-08-30:
ok. I followed now exactly that steps and got follow error:

/usr/bin/ld: cannot find -lSDL2
collect2: error: ld returned 1 exit status
Makefile:92: recipe for target 'default' failed
make: *** [default] Error 1

kisvegabor, 2018-08-31:
You just need to remove "-lSDL2" from the beginning of the makefile.

epikao, 2018-08-31:
ok. now it compiles without error and it works, thank you very much again :-)

epikao, 2018-09-03:
what does follow warning means?:
main.c:29:5: warning: implicit declaration of function ‘demo_create’ [-Wimplicit-function-declaration]
     demo_create();

kisvegabor, 2018-09-03:
I will update the framebuffer project and add the suggested updates including Makefile and evdev config.

> But I cannot see USB mouse or USB touch is working correctly with the demo- why that?
according  evtest, my usb-touch is using event 3 and I adapt that in the conf.h file.

Do the events arrive in `evdev_read`? Please, test with `printf`s.

> And is it normal that the demo changes every 1-2 seconds from "Write" Demo to "List" then to "Chart" etc.? 

`LV_DEMO_SLIDE_SHOW` enables it in `lv_ex_conf.h`.

> Also the "bubbles" picture is not visible on the keyboard demo?

The keyboard and the Text area should be opaque and bubbles should be visible. Is `LV_VDB_SIZE` > 0?

> BTW: what does follow warning means?:
main.c:29:5: warning: implicit declaration of function ‘demo_create’ [-Wimplicit-function-declaration]
demo_create();

You just need to include `lv_examples/lv_apps/demo/demo.h`

epikao, 2018-09-03:
The keyboard and the Text area should be opaque and bubbles should be visible. Is LV_VDB_SIZE > 0?

thanks, see below:

```
/* Horizontal and vertical resolution of the library.*/
#define LV_HOR_RES          (1024)
#define LV_VER_RES          (600)
#define LV_DPI              100

/* Size of VDB (Virtual Display Buffer: the internal graphics buffer).
 * Required for buffered drawing, opacity and anti-aliasing
 * VDB makes the double buffering, you don't need to deal with it!
 * Typical size: ~1/10 screen */
#define LV_VDB_SIZE         (20 * LV_HOR_RES)  /*Size of VDB in pixel count (1/10 screen size is good for first)*/
#define LV_VDB_ADR          0                  /*Place VDB to a specific address (e.g. in external RAM) (0: allocate automatically into RAM)*/
```

epikao, 2018-09-04:
I tested the touch again  see attached video.
Sometimes I get touch-inputs, but then it is very unsteady... 
[touch input(1).zip](https://github.com/littlevgl/lvgl/files/2347358/touch.input.1.zip)

The keyboard screen is still without bubbles and with USB mouse it is very difficult to control.
I don't know what that could be?

kisvegabor, 2018-09-05:
It's very strange that you see the "black list" and the "text area" on the same tab. The "list" should be on the "List" tab. It looks like this for me: [demo.mp4.zip](https://github.com/littlevgl/lvgl/files/2351892/demo.mp4.zip)

1. Have you done any modification in the lvgl or the demo?
2. Which branch do you use? It should be the same for lvgl and lv_examples too.  `master` or `dev-5.2`
3. In `lv_ex_conf` please set `LV_DEMO_WALLPAPER   0` and `LV_DEMO_SLIDE_SHOW   0`

Let's make it work first! After it, we can deal with the mouse.

epikao, 2018-09-05:
> Have you done any modification in the lvgl or the demo?
> Which branch do you use? It should be the same for lvgl and lv_examples too. master 

In demo and lvgl folder I didn't have done any modification.
I used exactly the branch which are linked here: http://www.vk3erw.com/index.php/16-software 
so means the branch "master".

> In lv_ex_conf please set LV_DEMO_WALLPAPER 0 and LV_DEMO_SLIDE_SHOW 0
both are already at 0

I suggest that there is something wrong in my main.c:
```
#include "lvgl/lvgl.h"
#include "lv_drivers/display/fbdev.h"
#include <unistd.h>
#include "lv_drivers/indev/evdev.h"
#include "lv_examples/lv_apps/demo/demo.h"

int main(void)
{
    /*LittlevGL init*/
    lv_init();

    evdev_init();
    lv_indev_drv_t indev_drv;
    lv_indev_drv_init(&indev_drv);
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.read = evdev_read;
    lv_indev_drv_register(&indev_drv);

    /*Linux frame buffer device init*/
    fbdev_init();

    /*Add a display the LittlevGL sing the frame buffer driver*/
    lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);
    disp_drv.disp_flush = fbdev_flush;      /*It flushes the internal graphical buffer to the frame buffer*/
    lv_disp_drv_register(&disp_drv);

    /*Create a "Hello world!" label*/
    lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);
    //lv_label_set_text(label, "Hello world!");
    demo_create();
    lv_obj_align(label, NULL, LV_ALIGN_CENTER, 0, 0);

    /*Handle LitlevGL tasks (tickless mode)*/
    while(1) {
        lv_tick_inc(5);
        lv_task_handler();
        usleep(5000);
    }

    return 0;
}
```

kisvegabor, 2018-09-05:
Your main.c looks fine. What if you enable `LV_DEMO_WALLPAPER        1`?
Please send an image or video.

epikao, 2018-09-05:
> Your main.c looks fine. What if you enable LV_DEMO_WALLPAPER 1?

now the bubbles are shown :-), mouse or touch input is still difficult.
Sometime I have Inputs, sometimes not....

kisvegabor, 2018-09-05:
And what about the list? Please send a short video as previously to be sure it's ok.

kisvegabor, 2018-09-05:
I see the image now above. It looks well.

So the touchpad:
Firstly we need to be sure the driver works well. So using `printf` in `evdev_read` please check what's going on. I don't know how your `evdev_read` looks like now, but you should add `printf`s like this:
```c
      struct input_event in;

    while(read(evdev_fd, &in, sizeof(struct input_event)) > 0) {
        if (in.type == EV_REL) {
            if (in.code == REL_X) {
                printf("REL_X %d\n", in_value);
                evdev_root_x += in.value;
            }
            else if (in.code == REL_Y) {
                printf("REL_Y %d\n", in_value);
                evdev_root_y += in.value;
            }
        } else if (in.type == EV_ABS) {
            if (in.code == ABS_X) {
                printf("ABS_X %d\n", in_value);
                evdev_root_x = in.value;
            }
            else if (in.code == ABS_Y) {
                printf("ABS_Y %d\n", in_value);
                evdev_root_y = in.value;
            }
        } else if (in.type == EV_KEY) {
            if (in.code == BTN_MOUSE || in.code == BTN_TOUCH) {
                if (in.value == 0)
                    evdev_button = LV_INDEV_STATE_REL;
                else if (in.value == 1)
                    evdev_button = LV_INDEV_STATE_PR;
            }
        } else {
            printf("EV_TYPE unknown\n", in_value);
        }
    }

    if (evdev_root_x < 0)
        evdev_root_x = 0;
    if (evdev_root_y < 0)
        evdev_root_y = 0;
    if (evdev_root_x >= LV_HOR_RES)
        evdev_root_x = LV_HOR_RES - 1;
    if (evdev_root_y >= LV_VER_RES)
        evdev_root_y = LV_VER_RES - 1;

    /*Store the collected data*/
    data->point.x = evdev_root_x;
    data->point.y = evdev_root_y;
    data->state = evdev_button;

    return false;
}
```

epikao, 2018-09-05:
if event0, mouse is working properly, also the demo change properly from list to write etc.
just I cannot see the mouse arrow so it's difficult... I will try now again with the touch...

kisvegabor, 2018-09-07:
Hi,

Look at e.g. this lines:
```
Event: time 1536152101.382872, type 3 (EV_ABS), code 0 (ABS_X), value 15423
Event: time 1536152101.382872, type 3 (EV_ABS), code 1 (ABS_Y), value 15127
```

The values are very big numbers. It seems you need to scale it down to be between `0 ... LV_HOR_RES` and `0 .. LV_VER_RES`.

So you need to figure out what are the minimum and maximum values from your mouse to do some math on them.

epikao, 2018-09-07:
> So you need to figure out what are the minimum and maximum values from your mouse to do some math on them.

Please note, is not the mouse, is my usb-touch.
The mouse is working properly now.

As I have assumed this values you mention are too high for your GUI.
But you must know that I use a typical USB Touch - and this value are usual and the touch is working properly with the raspbian desktop.
So maybe you have the possibilty to modify your GUI that it can also handle higher Values.

what do you think?

kisvegabor, 2018-09-07:
Sorry, I meant to say  "touch".

I think the scaling is handled by the TP driver of raspberry. So there are two options:
1. Somehow get the calibration data from the driver but I don't know if a generic solution exists. 
2. Add defines to `lv_ex_conf.h` to set the min and max coordinates and then the scaling can be done in the driver. The get the coordinates you can use the [tpcal](https://github.com/littlevgl/lv_examples/tree/master/lv_apps/tpcal) application or the log you already have.

epikao, 2018-09-07:
>Somehow get the calibration data from the driver but I don't know if a generic solution exists

I was able to change to get lower numbers, respectively between 0...1024. But I have still the same that Touch is not working.

> Add defines to lv_ex_conf.h to set the min and max coordinates and then the scaling can be done in the driver. The get the coordinates you can use the tpcal application or the log you already have.

can I change the Value here:?

[image]

kisvegabor, 2018-09-07:
Great, if you are in [0,1023] range then in the read function just do this:
```
last_x = (x_act * MONITOR_HOR_RES) / 1024;
last_y = (y_act * MONITOR_VER_RES) / 1024;
```
Where `x_act` and `y_act` are the curent coordinates.

For the point 2) I meant it's a possible feature. There no defines for it now. 

> would it be a 3rd possibilty to read the touch-data and use your function as below?
bool my_tp_read(lv_indev_data_t *data)

How do you mean it? `my_tp_read` is your current touchpad read function, isn't it?

epikao, 2018-09-07:
> How do you mean it? my_tp_read is your current touchpad read function, isn't it?

No as it is a USB-Touch I do not know how to read the touch-data and proceed to this function.

If it would be a I2C Touch I could read the Touch data and proceed to the function "my_tp_read(lv_indev_data_t *data)", but as it is a USB Touch I have at the moment no way to read the data via simple c code and proceed to your function program.

I was thinking that on Linux I can simply attach a USB-Touch like the USB Mouse and your GUI will recognize that.

thanks

kisvegabor, 2018-09-07:
I think it can be done like this (I'm not sure how you do it right now so I list all the steps):
1. Use `evdev.c` and `evdev.h` (or copy them)
2. Modify the `eved_read()` function to read from the touchpad's `evdev`
3. Once you have the coordinates scale and save them.

So I think it should work this way. Have you done this as well?

epikao, 2018-09-07:
no I didn't modify anything in evdev.c and evdev.h.

I will check and try to understand that.. thanks

epikao, 2018-09-07:
I now tried to replace the evdev.c code with your code above, but there is an undefined variable:
```
lv_drivers/indev/evdev.c: In function ‘evdev_read’:
lv_drivers/indev/evdev.c:75:39: error: ‘in_value’ undeclared (first use in this function)
                  printf("REL_X %d\n", in_value);
                                       ^~~~~~~~
lv_drivers/indev/evdev.c:75:39: note: each undeclared identifier is reported only once for each function it appears in
Makefile:88: recipe for target 'evdev.o' failed
```

which typ should I take for in_value?

kisvegabor, 2018-09-07:
It was just an example. 
Hopefully, you just need to scale the coordinates. If you are in [0,1024] range now try to replace the last lines with this:
```c
    /*Store the collected data*/
    data->point.x = (evdev_root_x * MONITOR_HOR_RES) / 1024;
    data->point.y = (evdev_root_y * MONITOR_VER_RES) / 1024;
    data->state = evdev_button;

    return false;
}
```

epikao, 2018-09-07:
I would like to see what your programs does exactly read.. so I think your code above with the "printf" could be helpful, but I'm still not able to printf some coordinate values on the screen. What should be the "in_value" for a typ-variable?

kisvegabor, 2018-09-07:
It should be `in.value`. Accidentally I mixed the `.` with `_`.

epikao, 2018-09-10:
> It should be in.value. Accidentally I mixed the . with _.

I cannot see any printf results on the screen, respectively how to do that without console? Can I start the console together with the demo?

kisvegabor, 2018-09-10:
You can start the demo in a console. If your output file is named "demo" (default in the Makefile) then `sudo ./demo` should work. It this case you should see the `printf`s in the terminal

epikao, 2018-09-10:
I will try tomorrow to start the demo directly in the console and not over SSH, I think that should solve the problem.

epikao, 2018-09-11:
unfortunately there are no data printed in the console, see attached picture
[image]

kisvegabor, 2018-09-12:
Even if you press the touchpad?

In the main while(1) add `printf("test\n")` to see  printf works.

epikao, 2018-09-13:
> In the main while(1) add printf("test\n") to see printf works.

I will test that asap.

I just recognized another warning from compiler, see below:

[image]

kisvegabor, 2018-09-13:
Really, it's a copy-past issue. That line should be `printf("EV_TYPE unknown\n")`

epikao, 2018-09-13:
with "printf" in main.c / while.. the result is as follow see below. So only a "p" letter is visible in the console.

kisvegabor, 2018-09-13:
It seems like a Raspberry specific issue. As the printing to the terminal is your best debug interface it would be quite important to make the "printf" work. As I don't know Raspberry, I don't know why it is not working. :(

However, from `evtest` log you can see the min and max values of the touchpad and use them to do the scaling.

Or see the sent event codes. Maybe it sends the scaled coordinates too. In this case, replace `ABS_X` and `ABS_Y` in `evdev.c` with the code of the scaled coordinates.

An other idea is to set `LV_INDEV_POINT_MARKER       3` in `lv_conf.h`. It should draw red points where the screen is pressed. For this you have to set `disp_drv.disp_fill = fbdev_fill;` when registering your display.

epikao, 2018-09-14:
ok. thankyou for your hints. 
if I exclude demo as follow //demo_create();, I can see the printf("test\n"); in my SSH console, but nothing about the printf from evdev.c.
If I exclude also the printf("test\n"); from main.c I see nothing more.

kisvegabor, 2018-09-14:
What if you add a printf to at the beginning of `evdev_read` like `printf("evdev_read called\n")`?

The other printfs should be printed when you press the touchpad.

epikao, 2018-09-14:
just found a mistake, event 0 in drv_conf.h was activated instead event 2, so didn't see anything because it always was requesting the mouse which was not connected... I can see the printf results from mouse in SSH.
I will update the printf results from event 2 later

epikao, 2018-09-14:
> For this you have to set disp_drv.disp_fill = fbdev_fill; when registering your display.

like follow?:

```

    /*Add a display the LittlevGL sing the frame buffer driver*/
    lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);
    disp_drv.disp_flush = fbdev_flush;      /*It flushes the internal graphical buffer to the frame buffer*/
    disp_drv.disp_fill = fbdev_fill;
    lv_disp_drv_register(&disp_drv);
```

kisvegabor, 2018-09-14:
Yes!

epikao, 2018-09-14:
and awesome, the demo is also working now :-)) , that's really a great GUI, tool :-) thank you again.

kisvegabor, 2018-09-14:
Cool! Have fun with LittelvGL! :)

closed 2018-09-14 by kisvegabor