lvgl/lvgl #390: [SDL PC simulator] How to redraw screen after regaining focus

carloantinarella, 2018-08-31:
Hi,
first of all thank you for developing such a beautiful and powerful library.
I'm developing a software for linux using the SDL library to be able to let the window manager to handle it as a normal window. I've seen that, if the window loses focus and gets partially or fully covered by other windows, when the SDL windows regains focus it doesn't automatically redraws its content.
Then, in the file "lv_drivers/display/monitor.c" I added the following lines at line 242:

```c
if ((&event)->type == SDL_WINDOWEVENT) {
            	switch ((&event)->window.event) {
            		case SDL_WINDOWEVENT_TAKE_FOCUS:
			            SDL_UpdateTexture(texture, NULL, tft_fb, MONITOR_HOR_RES * sizeof(uint32_t));
			            SDL_RenderClear(renderer);
			            SDL_RenderCopy(renderer, texture, NULL, NULL);
			            SDL_RenderPresent(renderer);
            			break;
            		default:
            			break;
            	}
            }
```
And it works absolutely fine for me. Are you interested to evaluate to officially implement a similar modification in your library?

Thank you
Carlo

carloantinarella, 2018-08-31:
I added one more line to my snippet of code:
```c
if ((&event)->type == SDL_WINDOWEVENT) {
            	switch ((&event)->window.event) {
            		case SDL_WINDOWEVENT_TAKE_FOCUS:
            		case SDL_WINDOWEVENT_EXPOSED:
			            SDL_UpdateTexture(texture, NULL, tft_fb, MONITOR_HOR_RES * sizeof(uint32_t));
			            SDL_RenderClear(renderer);
			            SDL_RenderCopy(renderer, texture, NULL, NULL);
			            SDL_RenderPresent(renderer);
            			break;
            		default:
            			break;
            	}
            }
```
In this case the window is redrawn also when (while remaining unfocused) is partially covered and then uncovered by other windows.

kisvegabor, 2018-09-03:
Hi,

 I'd happily add your suggestion! Can you create a Pull request?