lvgl/lvgl #391: Enable Right Mouse Click

renegat56, 2018-08-31:
Hi Gabor,

I got mouse working with own cursor image in linux framebuffer device. I can click left button to start some action. Now I want to extend it with an action with mouse right button click. Therefore I extended evdev_read() with BTN_RIGHT in.code and my own states like 

```c
bool evdev_read(lv_indev_data_t * data)
{
    struct input_event in;

    while(read(evdev_fd, &in, sizeof(struct input_event)) > 0) {
        if (in.type == EV_REL) {
            if (in.code == REL_X)
                evdev_root_x += in.value;
            else if (in.code == REL_Y)
                evdev_root_y += in.value;
        } else if (in.type == EV_ABS) {
            if (in.code == ABS_X)
                evdev_root_x = in.value;
            else if (in.code == ABS_Y)
                evdev_root_y = in.value;
        } else if (in.type == EV_KEY) {
            if (in.code == BTN_MOUSE || in.code == BTN_TOUCH) {
                if (in.value == 0)
                    evdev_button = LV_INDEV_STATE_REL; 	// released
                else if (in.value == 1)
                    evdev_button = LV_INDEV_STATE_PR;	// pressed (repeated in.value is 2)
            }
            if (in.code == BTN_RIGHT) {
                if (in.value == 0)
                    evdev_button = MY_INDEV_STATE_RIGHT_REL; 	// released
                else if (in.value == 1)
                    evdev_button = MY_INDEV_STATE_RIGHT_PR;	// pressed (repeated in.value is 2)
            }
        }
    }

    if (evdev_root_x < 0)
        evdev_root_x = 0;
    if (evdev_root_y < 0)
        evdev_root_y = 0;
    if (evdev_root_x >= LV_HOR_RES)
        evdev_root_x = LV_HOR_RES - 1;
    if (evdev_root_y >= LV_VER_RES)
        evdev_root_y = LV_VER_RES - 1;

    /*Store the collected data*/
    data->point.x = evdev_root_x;
    data->point.y = evdev_root_y;
    data->state = evdev_button;

    return false;
}

```

I also added my own states to lv_indev_state_t like

```c
/*States for input devices*/
typedef enum {
    LV_INDEV_STATE_REL = 0,
    LV_INDEV_STATE_PR,
	MY_INDEV_STATE_RIGHT_REL = 0,
	MY_INDEV_STATE_RIGHT_PR
}lv_indev_state_t;
```

But now I'm lost, the button reacts in the function 

```c
//Called when a button is released or long pressed
static lv_res_t btnm_action(lv_obj_t * btnm, const char *txt)
{
    printf("Button: %s released\n", txt);
    return LV_RES_OK; 				//Return OK because the button matrix is not deleted
}
```

on both left and right click on a button. How can I distinguish between BTN_LEFT and BTN_RIGHT event?
How can I access the state in data structure of evdev_read() and execute an action function for right mouse click independent of the lv button objects I have. Like show an fullscreen image by clicking left on a button in lv screen, but return to or redraw lv button screen with right click anywhere in image (all in /dev/fb0)

Thanks!

kisvegabor, 2018-09-03:
Hi,

I suggest using the `lv_indev_get_act()` function in the action. It tells you which indev triggered the action and you can decide what to do.

renegat56, 2018-09-13:
Thanks Gabor!

closed 2018-09-13 by renegat56