lvgl/lvgl #392: Website is blocking TOR users

ollieparanoid, 2018-09-03:
Hello there,

`littlevgl` seems to be a great library, and we are [evaluating to use it](https://gitlab.com/postmarketOS/pmbootstrap/issues/1657). I have noticed that the `littlevgl.com` website gives a 403 Forbidden error when accessing it with TOR. Is this on purpose, or could you maybe remove that restriction, at least for read-only content?

Thanks!

kisvegabor, 2018-09-03:
Hi,

it's not intentional. I googled on how to remove this restriction but didn't find anything useful. Can you give some hints how shall I search it?

kisvegabor, 2018-09-06:
Hi,

I ask the host but they don't know anything about it.

I'm using CDN to display the "cookie law". Is it problematic for Tor?

ollieparanoid, 2018-09-06:
Unfortunately still not working:

```html
<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>403 Forbidden</title>
</head><body>
<h1>Forbidden</h1>
<p>You don't have permission to access /
on this server.</p>
<p>Additionally, a 403 Forbidden
error was encountered while trying to use an ErrorDocument to handle the request.</p>
</body></html>
```

kisvegabor, 2018-09-07:
I also installed Tor and got the same error.

 From `Forbidden You don't have permission to access /` I suppose it's related to `.htaccess` but still I didn't find the root of this issue.

kisvegabor, 2018-09-07:
I find some info here: https://blog.torproject.org/comment/156175#comment-156175

I added an exit node to torrc: `EntryNodes {ca} StrictNodes 1` but it did't help.
Maybe the comment above tells more for you than me.

ollieparanoid, 2018-09-13:
What the comments over there are saying is basically, that somebody has probably sent spam from a Tor IP to the website in question, and therefore the website's admin has blocked that Tor IP. They suggested using another exit node, which means the IP visible to the website will be different. In the hope that another IP was not blocked by chance.

The Tor IPs are publicly known. It might also be that the admin has blocked all known Tor IPs.

However, your situation is a bit different since you are the admin, and from what you are saying, you did not block Tor on purpose. From what I understand, somewhere in the chain (either in your webserver config, or anything in front of it), has the Tor IPs in a blacklist that makes it return the 403 Forbidden page.

Looking at the network headers (F12 in most browsers), the 403 page seems to be coming from Apache. So it's probably in the Apache config somewhere?

kisvegabor, 2018-09-14:
Thank you for the explanation. I ask the host again.

kisvegabor, 2018-09-14:
I got an answer: they are not running "TOR proxy" therefore the website can nt be reash from TOR network. 

So are they simply not supporting TOR?

kisvegabor, 2018-09-19:
The technical support guy said that they are not blocking and IP by default but the servers have "certain" self-protection mechanisms. 

One of my friends also have a website at this host and his site doesn't work eighter from Tor.  

I'm afraid I can't do anything to fix it. :(

ollieparanoid, 2018-09-20:
> but the servers have "certain" self-protection mechanisms.

That sounds pretty much like the cause now.

> I'm afraid I can't do anything to fix it. :(

You have tried very hard though, thanks for the effort you have put into this!

Depending on how important this issue is to you, it seems the only thing you could do is considering to switch the hosting company if the current one is really not willing to fix this.

(I am assuming that the "hosting company" is controlling the Apache setup, and you are only providing the HTML pages or something like that.)

kisvegabor, 2018-09-21:
Because I have no problems with the host so far I wouldn't like to change now. Sorry. :( 
If this issue will be relevant for more people I will consider moving to a new host. 

As there is nothing to do now I close this issue.
Thank you for pointing out this issue and helping to figure out the cause of it.

closed 2018-09-21 by kisvegabor