lvgl/lvgl #393: new object "knob"

ripe909, 2018-09-03:
Hello,

As I learn lvgl I have been building some of my own custom objects, including a knob, which I based on the Gauge object (a knob is actually simpler than the gauge).

I think this would be a useful standard object type.  

I would also like to see some additional features I wasn't successful in adding to my object (yet):  
* The ability to define a label above or below the knob to show the current value.
* The ability to click on and then drag up/down to set the knob value. 
* The ability to set the value range

I used a circle for the knob indicator since you don't have to worry about rotating the indicator this way.

Here is a simple screen shot:
[image]

kisvegabor, 2018-09-05:
Hi,

It's a very good idea to add knob object! And I'm very happy to see that you are interested in adding new objects! :)

> The ability to define a label above or below the knob to show the current value.

As the objects can draw on the area of their size you can't add a label simply below it. I suggest to add the label in the middle of the knob or add it as another label object.

> The ability to click on and then drag up/down to set the knob value.

I suggest using the LV_SIGNAL_PRESSING signal in the signal function. 

> The ability to set the value range (mapped to the rotation range)

You just need to calculate the ratio between `angle range`, `current angle` and `value range`, `value`. 

I also have two question:
1. "based on the Gauge object" means "knob" is similar to gauge or it's derived from it. Derived means you create a gauge object and extend it (like `lv_btn` extends `lv_cont`).
2. What do you think about an encoder option where the current value is incremented/decremented on rotate? Can it be a config option to select between the normal potentiometer and encoder mode?

 Do you plan to share your work? It would easier to help. For example I could show how to use the `LV_SIGNAL_PRESSING` in your code.

kisvegabor, 2018-10-07:
As there was no activity here for a while I close this issue.

If you have remarks about this topic feel free to comment here and reopen the issue.

kisvegabor, 2018-11-29:
There was no activity here fora  while so I close this issue.
I'm still looking forward to the Pull request.  :)

closed 2018-11-29 by kisvegabor

ripe909, 2018-11-29:
I have a commit for the knob object, but I’m not sure how to issue the pull request...

embeddedt, 2018-11-30:
@ripe909 You need to create a fork of LittlevGL and apply your changes to it. Then you can use the GitHub website to create the pull request from your fork to `littlevgl/lvgl:dev-5.3`.

@kisvegabor Since the comment of @ripe909 suggests the object is ready maybe we should reopen this issue.

embeddedt, 2018-11-30:
@kisvegabor See #593.

kisvegabor, 2018-11-30:
@ripe909 @embeddedt 
Thank you, lets continue the discussion in the PR!