lvgl/lvgl #394: Multiple screen question

rbultman, 2018-09-04:
What is the best approach for creating multiple screens and switching between those screens without using the tabbed interface?

Should I start with the base object or something else, like a container, and then use the hidden attribute to hide/reveal each screen?

Thanks.

rbultman, 2018-09-04:
So, I tried creating two new screens using `lv_obj_create(NULL, NULL)`, but my system crashes when I try to hide a screen since the hide attempts to call `signal_func` on the parent, but there is no parent.  Either what I am doing is "wrong" conceptually or the `lv_obj_set_hidden` function should check for a null pointer before dereferencing in the last line of the function.

kisvegabor, 2018-09-05:
Hi,

If you have two screens you can change between them with `lv_scr_load(new_scr)`.

For example:
```c
lv_obj_t * scr1  = lv_obj_create(NULL, NULL);
lv_obj_t * scr2  = lv_obj_create(NULL, NULL);
lv_scr_load(scr1);

...

lv_scr_load(scr2);
```

rbultman, 2018-09-05:
Well dang.  Pretty sure I tried that before and it didn't work.  I just tried it again and it worked.  I was messing with hidden too, so maybe there was some other interaction there that I don't understand.  Anyhow, thanks for the fast reply.

closed 2018-09-05 by rbultman

kisvegabor, 2018-09-05:
Your welcome!

Always there is one active screen so you can't hide it else it wouldn't be anything to display.